fritzing/fritzing-app #3043: Circuit traces disconnect and scatter constantly.

Justnthenick, 2015-01-08:
I save my circuit alot and I leave it open on my MacBook 2012 13 inch. I come back to it and the traces are all messed up. Some have skips in the traces and others arent connected, and just the other night, everything was done. This happens on every computer I own, all macs.

KLiFF2606, 2015-11-27:
I have the same issue on Ubuntu 14. When I reopen a file the traces are some how misinterpreted: the bend-points order is not read correctly. Here is an example of a trace with 2 bend-points:
- Before saving and closing the file the order is: A-B-C-D.
- After reopening the file: A-C-B-D.

I hope this helps to understand the bug.

aknoerig, 2015-11-30:
Duplicate of #3115. Thanks for the additional insight!

closed 2015-11-30 by aknoerig