fritzing/fritzing-app #3044: Cannot resize to monitor - too big

simon-Br, 2015-01-14:
I cannot resize the window to fit my monitor which is 1024x768 resolution. The window sizes are too large. Is there a way to reduce these to fit correctly?

Thanks,
Simon.

el-j, 2015-01-28:
hello simon, 
on this small resolution you could only release the pallets to solve this problem. just track them out and i should work. if you make fritzing full-screen it should work.

we know about the "minimal"-width problem but cannot fix it because we need to reserve space for the bottom-tool-bar. 

[image]

closed 2015-01-28 by el-j

simon-Br, 2015-01-28:
> image

Hi,

What do you mean by "small" resolution? This is the maximum that this monitor is capable of
and any higher, it would also be impossible to read anything.

Thanks,
Simon.

chrisyokum, 2015-01-28:
1024x768 is low resolution, it is not very common especially in engineering. I think that's what he means.

grince, 2017-12-13:
I have the same problem with Fritzing on my Debian Stable (9.x) with a Full-HD Resolution.
I am not sure what goes wrong, but this is a screenshot of the app as I start it and I'll have an xterm below to show the size of a "normal" xterm. I have no idea why this happens, is there a way to have a smaller fontsize or something similar?

[image]

juanmitaboada, 2019-01-04:
Having same problem here in Debian 9. My resolution is 1920x1080 and everything looks so big I can not fit the window either into my desktop.

AugustoCiuffoletti, 2022-01-09:
@juanmitaboada your solution works on Ubuntu 20.04.3 LTS and Fritzing 0.9.3b.
Many thanks