fritzing/fritzing-app #3048: Segmentation fault when running Autoroute

zhuowei, 2015-01-26:
I tried autorouting a circuit for the first time, and Fritzing crashed.

I'm using fritzing.0.9.1b.64.pc on Windows 8.1.

Crash log from Windbg:
```
Microsoft (R) Windows Debugger Version 6.3.9600.16384 AMD64
Copyright (c) Microsoft Corporation. All rights reserved.

*** wait with pending attach

************* Symbol Path validation summary **************
Response                         Time (ms)     Location
Deferred                                       srv*d:\symbols*http://msdl.microsoft.com/download/symbols
Symbol search path is: srv*d:\symbols*http://msdl.microsoft.com/download/symbols
Executable search path is: 
ModLoad: 00007ff6`dfb10000 00007ff6`e02c4000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\Fritzing.exe
ModLoad: 00007ffd`88020000 00007ffd`881cc000   C:\WINDOWS\SYSTEM32\ntdll.dll
ModLoad: 00007ffd`85ee0000 00007ffd`8601e000   C:\WINDOWS\system32\KERNEL32.DLL
ModLoad: 00007ffd`854d0000 00007ffd`855e5000   C:\WINDOWS\system32\KERNELBASE.dll
ModLoad: 00007ffd`66a20000 00007ffd`66a68000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\Qt5PrintSupport.dll
ModLoad: 00000000`70d70000 00000000`70db3000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\Qt5Svg.dll
ModLoad: 00000000`70860000 00000000`70d6c000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\Qt5Widgets.dll
ModLoad: 00007ffd`72600000 00007ffd`72615000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\Qt5SerialPort.dll
ModLoad: 00000000`70820000 00000000`70854000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\Qt5Sql.dll
ModLoad: 00000000`70720000 00000000`7081d000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\Qt5Network.dll
ModLoad: 00000000`706e0000 00000000`70716000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\Qt5Xml.dll
ModLoad: 00007ffd`58320000 00007ffd`586c9000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\Qt5Gui.dll
ModLoad: 00000000`70240000 00000000`706de000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\Qt5Core.dll
ModLoad: 00007ffd`60f10000 00007ffd`60fb7000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\MSVCP110.dll
ModLoad: 00007ffd`60e30000 00007ffd`60f04000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\MSVCR110.dll
ModLoad: 00007ffd`87970000 00007ffd`87ac1000   C:\WINDOWS\system32\GDI32.dll
ModLoad: 00007ffd`87dd0000 00007ffd`87f47000   C:\WINDOWS\system32\USER32.dll
ModLoad: 00007ffd`7dff0000 00007ffd`7e072000   C:\WINDOWS\SYSTEM32\WINSPOOL.DRV
ModLoad: 00007ffd`87f50000 00007ffd`88006000   C:\WINDOWS\system32\COMDLG32.dll
ModLoad: 00007ffd`86040000 00007ffd`87558000   C:\WINDOWS\system32\SHELL32.dll
ModLoad: 00007ffd`87640000 00007ffd`8781a000   C:\WINDOWS\system32\SETUPAPI.dll
ModLoad: 00007ffd`87b70000 00007ffd`87c1a000   C:\WINDOWS\system32\ADVAPI32.dll
ModLoad: 00007ffd`859f0000 00007ffd`85a4a000   C:\WINDOWS\system32\WS2_32.dll
ModLoad: 00007ffd`84850000 00007ffd`848f4000   C:\WINDOWS\SYSTEM32\DNSAPI.dll
ModLoad: 00007ffd`85240000 00007ffd`8541f000   C:\WINDOWS\system32\CRYPT32.dll
ModLoad: 00007ffd`59d90000 00007ffd`59e66000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\libGLESv2.dll
ModLoad: 00007ffd`87c30000 00007ffd`87dc4000   C:\WINDOWS\system32\ole32.dll
ModLoad: 00000000`70010000 00000000`70238000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\icuin51.dll
ModLoad: 00000000`6fe60000 00000000`70004000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\icuuc51.dll
ModLoad: 00007ffd`85e20000 00007ffd`85eca000   C:\WINDOWS\system32\msvcrt.dll
ModLoad: 00007ffd`85760000 00007ffd`857b4000   C:\WINDOWS\system32\SHLWAPI.dll
ModLoad: 00007ffd`6aa40000 00007ffd`6aae4000   C:\WINDOWS\WinSxS\amd64_microsoft.windows.common-controls_6595b64144ccf1df_5.82.9600.17415_none_34aa3313958e7a52\COMCTL32.dll
ModLoad: 00007ffd`85a50000 00007ffd`85c61000   C:\WINDOWS\SYSTEM32\combase.dll
ModLoad: 00007ffd`855f0000 00007ffd`8563f000   C:\WINDOWS\system32\CFGMGR32.dll
ModLoad: 00007ffd`87820000 00007ffd`87961000   C:\WINDOWS\system32\RPCRT4.dll
ModLoad: 00007ffd`87b10000 00007ffd`87b69000   C:\WINDOWS\SYSTEM32\sechost.dll
ModLoad: 00007ffd`87c20000 00007ffd`87c29000   C:\WINDOWS\system32\NSI.dll
ModLoad: 00007ffd`85220000 00007ffd`85231000   C:\WINDOWS\system32\MSASN1.dll
ModLoad: 00007ffd`740f0000 00007ffd`74311000   C:\WINDOWS\SYSTEM32\d3d9.dll
ModLoad: 00000000`6e910000 00000000`6fe5a000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\icudt51.dll
ModLoad: 00007ffd`7c980000 00007ffd`7c98a000   C:\WINDOWS\SYSTEM32\VERSION.dll
ModLoad: 00007ffd`82f70000 00007ffd`82f91000   C:\WINDOWS\SYSTEM32\dwmapi.dll
ModLoad: 00007ffd`82da0000 00007ffd`82e52000   C:\WINDOWS\SYSTEM32\SHCORE.DLL
ModLoad: 00007ffd`87ad0000 00007ffd`87b06000   C:\WINDOWS\system32\IMM32.DLL
ModLoad: 00007ffd`857c0000 00007ffd`85913000   C:\WINDOWS\system32\MSCTF.dll
ModLoad: 00007ffd`850b0000 00007ffd`850bb000   C:\WINDOWS\SYSTEM32\CRYPTBASE.DLL
ModLoad: 00007ffd`85040000 00007ffd`850a3000   C:\WINDOWS\SYSTEM32\bcryptPrimitives.dll
ModLoad: 00007ffd`58210000 00007ffd`5831c000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\platforms\qwindows.dll
ModLoad: 00007ffd`7ddd0000 00007ffd`7ddf2000   C:\WINDOWS\SYSTEM32\WINMM.dll
ModLoad: 00007ffd`87570000 00007ffd`87631000   C:\WINDOWS\system32\OLEAUT32.dll
ModLoad: 00007ffd`6a330000 00007ffd`6a342000   C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\libEGL.dll
ModLoad: 00007ffd`7dd90000 00007ffd`7ddba000   C:\WINDOWS\SYSTEM32\WINMMBASE.dll
ModLoad: 00007ffd`83f60000 00007ffd`83f88000   C:\WINDOWS\SYSTEM32\DEVOBJ.dll
ModLoad: 00007ffd`83fd0000 00007ffd`83fdb000   C:\WINDOWS\SYSTEM32\kernel.appcore.dll
[... 62 lines cut ...]
000000a1`25a4c620 00007ffd`87dd2367 USER32!UserCallWinProcCheckWow+0x149
000000a1`25a4c6f0 00000000`7044d0cc USER32!DispatchMessageWorker+0x1a7
*** WARNING: Unable to verify checksum for C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\platforms\qwindows.dll
*** ERROR: Symbol file could not be found.  Defaulted to export symbols for C:\Users\zhuowei\Documents\winprogress\fritzing.0.9.1b.64.pc\platforms\qwindows.dll - 
000000a1`25a4c770 00007ffd`5823031d Qt5Core!QEventDispatcherWin32::processEvents+0x4cc
000000a1`25a4fad0 00000000`7040a5b1 qwindows!qt_plugin_query_metadata+0x1e9ed
000000a1`25a4fb40 00000000`7040c647 Qt5Core!QEventLoop::exec+0x1c1
000000a1`25a4fbc0 00007ff6`dfb31bd7 Qt5Core!QCoreApplication::exec+0x187
000000a1`25a4fc30 00007ff6`dfdca5ce Fritzing+0x21bd7
000000a1`25a4fca0 00007ff6`dfdc9f60 Fritzing+0x2ba5ce
000000a1`25a4fd10 00007ffd`85ee13d2 Fritzing+0x2b9f60
000000a1`25a4fd50 00007ffd`880903c4 KERNEL32!BaseThreadInitThunk+0x22
000000a1`25a4fd80 00000000`00000000 ntdll!RtlUserThreadStart+0x34
0:000> r
rax=000000a140d41020 rbx=0000000000000028 rcx=fffffffffffffffd
rdx=ffffffffffffff9d rsi=0000000000000028 rdi=0000000000000048
rip=00007ff6dfcdfb97 rsp=000000a125a499f0 rbp=000000a12f024c60
 r8=fffffffffffffef4  r9=00000000fffffee0 r10=ffffffffffffffff
r11=000000000000000f r12=000000000000002d r13=000000a125a49ae8
r14=ffffffffffffffff r15=000000a134423790
iopl=0         nv up ei ng nz na pe nc
cs=0033  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00010280
Fritzing+0x1cfb97:
00007ff6`dfcdfb97 48890cd0        mov     qword ptr [rax+rdx*8],rcx ds:000000a1`40d40d08=????????????????
```

zhuowei, 2015-01-26:
I uploaded the file that causes the crash: https://drive.google.com/file/d/0B-fqnmBSesbtWFh4OFZkd0I0XzA/view?usp=sharing

KjellMorgenstern, 2019-07-18:
Copied files from google docs to github:
[motordriver.fzz.zip](https://github.com/fritzing/fritzing-app/files/3407326/motordriver.fzz.zip)

failiz, 2021-02-14:
I could not replicate this in 0.9.4.

KjellMorgenstern, 2021-02-14:
Several crashes were fixed for 0.9.4 I had initially even removed the autoroutes, since it was easy to crash. But several bug were fixe, and we could keep the autorouter in.

closed 2021-02-14 by KjellMorgenstern