fritzing/fritzing-app #3049: Save as in code view

jokohoko, 2015-01-26:
When working in the code view i can't save it at an other location.

Even saving the fritzing design does not copy the ino file along with it.

Also it would be great is the code would be saved inside a folder with the same name the way arduino likes it. So even if you have to use the code seperatly it still works.

arvindpdmn, 2015-11-21:
In a related issue, I found that when creating code for first time and saving it as a new ino file works. But if that ino file already exists in folder, I cannot overwrite it. There is a prompt to overwrite but it doesn't work as expected.

KjellMorgenstern, 2022-01-12:
The file can be renamed like this:
[image]

The dialog then also allows saving the file to a different folder.

Closing because it seems the issue does not occur anymore.

closed 2022-01-12 by KjellMorgenstern