fritzing/fritzing-app #3054: Position of dropdown on Mac OS X Yosemite

ronaldlokers, 2015-02-12:
When I click on a item that opens a dropdown I get some unexpected behaviour. The dropdown will be opened on another position then where I clicked.

This problem prevents me to go fullscreen because some popups will else open outside of the window.

Please let me know if you need any extra information to be able to reproduce this problem.

[image]

ronaldlokers, 2015-02-16:
I discovered that I don't have this problem when I use just 1 display, it only happens when I have 2 displays. So I think this is the same problem as in #3036

But the white text color in the popup is also there when using just 1 color, so that is another bug.

failiz, 2021-03-12:
This is a duplicate of #3036
Not sure about the white colour, but I think that it has been addressed.

closed 2022-11-16 by KjellMorgenstern