fritzing/fritzing-app #3060: Transistors leg order can not be changed

EeroKH, 2015-02-28:
I need to use transistor 2N3904 (pin order is 1 Emitter, 2 Base 3 Collector)
In all available transistors the order is 1 Base, 2 Emitter, 3 collector and they do not change no matte what transistor I select or what order I select to the pins. (In some earlier versions of Fritzing this has worked)

I use Fritzing version 0.9.1 in Linux Mint

el-j, 2015-04-17:
did you try the npn (ebc) from the core? all versions are available.
[image]

EeroKH, 2015-04-17:
That seems to work .
and I can not rebuild the situation.

If it was not this:
- you can not select BEC, BCE and EBC.
I just think I had to put a component in  |:) even the component layout
said (:|
But I can no more remember what I was doing so I can not verify.

Next time I must be more specific (tell how to get the situation).

Thank You
- Eero -

chrisspen, 2016-02-25:
This is also a problem with mosfets...except none of the included parts have an option to specify pin ordering. They all go GDS but most datasheets I see order them DGS.

When editing a part, how do you change the pin numbers? They don't appear to be editable.

EeroKH, 2016-02-25:
In PCB the wireing goes ok. it changes as ok when changing the pin order
but
in Schematic the pin numbers do not change when chancing the pin order in
transistor.

here bellow datasheet of
Normal signal NPN transistor and
High voltages NPNtransistor with different pin order

Notice the case numbering is same in both cases but the signal name chances
but in Frizzing the case numbering changes when signal name changes

This caused some confusion until I noticed this

Thank You
- Eero -

[image: Tekstiin lisättävä kuva 1]

[image: Tekstiin lisättävä kuva 2]

failiz, 2020-12-29:
The only issue reported here is that you cannot change the pin order in mosfets, but this does not require any coding. It is just to add the mosfets as different parts as in the bipolar transistors case. Move to the part repository?

KjellMorgenstern, 2020-12-29:
Related to #2849 . Lets first look if we can add special parts, instead of creating all possible combinations in the parts repo

failiz, 2021-01-03:
I created issue #3775 to leave it in the app database. So, we can move this issue to the part repository as workaround until we implement #3775.