fritzing/fritzing-app #3061: Developer Wiki: Sketch File Format. How does "transform" work?

htho, 2015-03-04:
Hi,
This is some kind of a developer-documentation related "issue":
The Wiki explains the file-format. But it does not describe how the "transform" works. There are those strange xml attributes: m11, m12, ..., m33 What do they mean?
https://github.com/fritzing/fritzing-app/wiki/2.2-Sketch-file-format
I'd really like to understand the whole positioning stuff.

Some Background:
Id like to be able to create a frontplate for a casing I will 3D-Print. The casing will be designed in OpenSCAD, but I need to know where to place the holes for LEDs and Buttons. It should not be too hard to parse the .fz file to get position and orientation of the parts that need holes in the case. Then I can fit this (very simple) 3D model of the PCB into the case and have it printed as I need it. If I am successfull I will put my tool on github ;-).

aknoerig, 2015-03-04:
Frontplates, yummy!

m11, m12 etc. simply define an SVG transform matrix (3x3). Good old high-school mathematics that seems difficult at first, but is actually pretty simple and luckily very standard. We're using it for optional rotaion/flipping of elements.

htho, 2015-03-12:
Hi,
it took a while to work out the matrix stuff as these are 3x3 QTransform matrices and OpenSCAD has 4x4 matrices. But hey it works now. Thanks!
Maybe youd like to have a look at my Project:
https://github.com/htho/fzz2scad

aknoerig, 2015-03-13:
Nice, sounds really promising! Can you post some screenshots of an example? Then I'm happy to put it on our blog. :)

htho, 2015-03-13:
Hi, I have added two screenshots to the repo.
I'd be happy if you'd put it on your blog!

failiz, 2020-10-11:
Should we close this issue? The question has been replied and I am not sure if it is necessary to specify this in the wiki.

closed 2021-01-19 by htho

failiz, 2021-01-19:
Thanks for closing the issue. Added to the wiki.