fritzing/fritzing-app #3063: behaviors making fritzing unuseable (duplicate, part overlapping, schematic/code interaction)

madias123, 2015-03-12:
Sadly there are some major issues, making fritzing nearly unuseable - here is my personal top list:
1. duplicate/copy than paste:
the position of the component after paste is anywhere but not on mouse pointer, mostly there, where even the most components are and now we have to "play" with the
2. overlapping parts
it is difficult and almost impossible to "click" the part you want if they are overlapping (best thing: IC on top and a SMD resistor on the bottom, try to catch the resistor! - no way)
3. rectangular PCB 
the pcb should be invisable while choosing multiple components. The could be an addon for the "locked" status
4. Interaction between schematic and PCB view (I don't use breadboad)
The placing of the components in PCB set in schematic (and vice versa) are completly annoying. Worst thing, and now fritzing is unuseable: make a complex sub circuit (say a opamp with resistor, caps,...) set it up in both: PCB and schematic. Now copy the sub circuit, because you need it more than one time. Now you can choose where it is completely messed up in PCB or schematics. There is no chance to bring it back to a right order.
5. values of resistors, caps, elkos....
Why is there a (short) list of values? This is annoying and not neccessary. Just let us enter the right value.
6. sending components to the back:
Silk screen should be mirrored.
7. auto routing:
Pardon, but this doesn't function, not even on small circuits. But for real: It's better to draw it by hand (even in eagle...)
This is only a summery of the most annoying and disfunctional things I found out. I've done a bigger project (quad analog filter on a 10x10 board) and I got mad and frustrated.

I do not know which group of people for this program is made:
Beginners: ok, maybe. Putting 2-3 resistors and caps together on a ARDUINO "shield" 
Intermediate: See my list.
Profis: nop not the target group, they use eagle or something else.

So sorry for my rude words, but they are at least honest.

claudiosousa, 2015-03-12:
Very good feedback, I suffer from the same pains. Is completely worth
fixing all these issues. Fritzing useability is great! Please fix all the
bugs !

otede, 2015-08-15:
Ad. 5.
Fritzing caters to hobbysts hence typical values are useful. You can enter any desired value.

You don't need to be intermediate to find Fritzing lackluster. I'm doing a simple PC fan controller and there are lots of problems with it. Lack of very basic parts, multitude of specialized parts no one knows exist if not using sparkfun store, creation of bogus unwanted connections/short-circuits on perfboard...

KjellMorgenstern, 2019-11-25:
Created #3568 for 1.

KjellMorgenstern, 2019-11-25:
Created #3569 for 2.

KjellMorgenstern, 2019-11-25:
Created #3570 for 4.

The other issues are unclear, already implemented or not helpful. Closing.

closed 2019-11-25 by KjellMorgenstern