fritzing/fritzing-app #3064: uploading sketches or parts to github - adding feature to fritzing app

aquilanorica, 2015-03-16:
Hello fritzing team,

it would be a nice feature being able to upload sketches or parts to github for contribution right from the fritzing app such as "publish online".
Is there a recommended way or description to do so?

Thanks for your good work.

Keep on going.

Aquila

failiz, 2021-03-13:
duplicate of #3752 and #835?

KjellMorgenstern, 2021-03-14:
Yes, uploading sketches is #3752 and uploading parts is #835 . However, #835 only looks at the github toolchain. A huge number of users are not familiar with github, so they just upload parts to the forum, where they are eventually found or even kind of verified. 
I'll change this ticket to be about providing a way of easily creating a PR from a new part within Fritzing.

KjellMorgenstern, 2021-03-14:
Add an option to the parts editor to create a pull request