fritzing/fritzing-app #3065: Developer Wiki: Sketch File Format. Views and their Layers.

htho, 2015-03-17:
Hi,

I need to find out if a part should be shown in the PCB View.
Until now I simply selected parts that have `./instances/instance/views/pcbView/geometry[@z > 0]` (z > 0 in pcbView). But this does not do the trick:
```xml
<instance moduleIdRef="20.28PerfboardModuleID" modelIndex="85495141" path="/home/htho/.config/Fritzing/partfactory/1916e24579d024323d346bbeacd0733d/core/20.28PerfboardModuleID.fzp">
    <property name="size" value="20.28"/>
    <title>Perfboard1</title>
    <views>
        <breadboardView layer="breadboardbreadboard">
            <geometry z="1.50003" x="-36" y="-144"/>
            <connectors>
                <!-- [...] -->
            </connectors>
        </breadboardView>
        <pcbView layer="breadboardbreadboard">
            <geometry z="2.50002" x="-143.983" y="147.941"/>
            <connectors>
                <!-- [...] -->
            </connectors>
        </pcbView>
        <schematicView layer="breadboardbreadboard">
            <geometry z="2.50003" x="-77.2891" y="74.8847"/>
            <connectors>
                <!-- [...] -->
            </connectors>
        </schematicView>
    </views>
</instance>
```
This perfboard should not be in the pcbView but `z > 0`!

Anyway, I thought `./instances/instance/views/pcbView[@layer="copper0"]` might help, but again this perfboard confuses me. I thought `./breadboardView[@layer="breadboard"]` would put that part in the breadboardView. But this perfboard has a layer named `breadboardbreadboard`.
So, what is it? How do I know if a part should be represented in the PCB View?

failiz, 2020-11-30:
This is more a question for the forum than a bug report. Please, check this forum thread:
https://forum.fritzing.org/t/breadboard-versus-breadboardbreadboard-layer/10445
Maybe we should document this in the wiki.

htho, 2021-01-19:
@failiz thats what the wiki is for ;-)

closed 2021-01-19 by htho

failiz, 2021-01-19:
I added documented the use of breadboardbreadboard  id in the wiki. Thanks for closing the issue.
By the way, if you do not want the schematic or pcb view, do not include them in the part definition.