fritzing/fritzing-app #3066: Sparkfun's 10mm LED icon problem in Fritzing Breadboard view

Jasmohan-ThandaLe, 2015-03-24:
Sparkfun's 10mm LED icon problem in Fritzing Breadboard view

Here is a Red 3mm LED next to a 10mm LED (Sparkfun) in breadboard view

[image]

Before I explain my problem let's go over some basics:-

A diode has two terminals. The positive side is called the anode, and the negative one is called the cathode.
The diode circuit symbol, with the anode and cathode marked.
[image]

LED stands for light-emitting diode, which means that much like their diode cousins, they’re polarized. There are a handful of identifiers for finding the positive and negative pins on an LED. You can try to find the longer leg, which should indicate the positive, anode pin.
(Source: https://learn.sparkfun.com/tutorials/polarity/diode-and-led-polarity)

Parts of an LED
[image]
(Source: http://en.wikipedia.org/wiki/Light-emitting_diode)

We can identify the polarity of an LED by looking inside it's casing. The larger end inside the led is -ve and the shorter one is +ve, that is how we find out the polarity of a LED.
[image]
(Source: http://www.instructables.com/id/IDENTIFY-THE-ANODECATHODE-of-LEDs/)

Now that you understand this much, I shall explain the problem.

If you look closely in the LED icons in the Breadboard view of Fritzing, we should be able to identify the positive and negative terminal easily by looking inside the casing.
i.e. Bigger element is Cathode and Smaller element is Anode.

[image]

So, looking at the icons as reference, I have connected the (presumably) Anodes and Cathodes of the 2 LEDs together by wires. But if we see the schematic view, the problem becomes clear.

[image]

The anode and cathode pins on the 10mm LED from sparkfun are reversed.

Since this is a part in the default library of Fritzing, and all installations have the same icons; I request that this problem be rectified so that new comers and experienced users don't get confused.

Here is some more information about the 10mm LED part:-
[image]

I may also be wrong about this, so if anyone can confirm this then that will be great.

One solution for the time being is to create a new part with the corrections; but if it gets fixed by fritzing in a software update then it is even better. That way, the problem may be solved for all the users.

el-j, 2015-04-17:
thanks for the bug report. you are right. the wrong part is now "obsolete". so the user get the hint to change to part to the new version.

closed 2015-04-17 by el-j

Jasmohan-ThandaLe, 2015-04-17:
el-j Thank you for your time.

commit 65b17d14 (2015-04-17): fixed #3066 - sparkfun10mm led