fritzing/fritzing-app #3073: Export to png does not update until restart fritzing and reload file.

SzymonZy, 2015-04-14:
I jut notice that if you export schematic to PNG, change something, export again ... output fill will not include changes made since last export. 

You have to turn off fritzing, reload file and export again. That will work.

thomasw-mitutoyo-ctl, 2020-08-28:
Still the same behavior in 0.9.4,

failiz, 2020-10-11:
Could you provide with detailed steps to reproduce it?
I can only reproduce it (0.9.4 in windows) when changing the name of a resistor. Steps to reproduce it:
1. Add a resistor in the schematic view of a new file 
2. Export to png
3. Double click in R1, set label to R2 and click ok
4. Export to png and the image still says R1

The name of the resistor updates only after: adding a second resistor, draw a wire between the resistors and move the first resistor.

thomasw-mitutoyo-ctl, 2020-10-12:
Steps for me:

0. Create a new untitled sketch
1. Add a resistor and have it selected (see the selection rectangle)
2. Export as PNG. Exported PNG will have the resistor selected
3. Select the breadboard (selection rectangle changes)
4. Export as PNG. Confirm overwriting the existing image or choose a different file name

Result: the PNG file still has the resistor selected

[image]

thomasw-mitutoyo-ctl, 2020-10-12:
Related code: 

mainwindow_export.cpp, line 970: creation of the button, associate with doExport()
mainwindow_export.cpp, line 481: doExport() method
mainwindow_export.cpp, line 548: calls exportAux()
mainwindow_export.cpp, line 556: exportAux() method

In sketchwidget.cpp, I find a lot of  ```// TODO: replace scene()->items()```. Why should it be replaced? Does it affect export as well? Should it be replaced there, too?

KjellMorgenstern, 2020-10-18:
Fixed by PR #3723 . Note that the export method still can not deal with the selection. It might still use a dangling pointer or at least a copy of the selection that should have been cleaned. The selection just doesn't reach the faulty part of the code anymore because the fix temporarily deselects.

failiz, 2021-03-18:
already released, close

closed 2021-03-19 by KjellMorgenstern