fritzing/fritzing-app #3075: Grid alignment (at least) for the four pin pushbutton changed.

htho, 2015-04-15:
When opening files created with Fritzing 0.9.1b using Version 0.9.2b, at least for the four pin pushbutton, the pin that is aligned to the grid changed from pin 1 to pin 2. When adding new buttons all the existing buttons need to be rotated 180° first as new buttons cant be aligned correctly.
Here is a file that was created using Version 0.9.1b: https://github.com/htho/fzz2scad/blob/master/testing/pushbutton_4pin_on_10x10mm.fzz

Until now I assumed pin1 is always the pin that is alligned to the grid. If that is not the case, how is it determined?

el-j, 2015-04-16:
this is very strange. there was no change during fritzing 0.9.0b.

htho, 2015-04-16:
Could you replicate this behaviour?

el-j, 2015-04-16:
yes i can, but i cannot find out why this is happening

htho, 2015-04-21:
If I correctly understand [`findAlignmentAnchor()`](https://github.com/fritzing/fritzing-app/blob/2feee8a3707d4cda887e3899597b8f85f601a38f/src/sketch/sketchwidget.cpp#L2537), parts/items can either be aligned at their top-left corner or to their center?

htho, 2015-04-22:
The aligned pin is not always pin2 sometimes it is pin4. But I cant reproduce this behaviour.

htho, 2015-04-22:
I've been looking deeper into `cachedConnectorItems()` and found it's [definition](https://github.com/fritzing/fritzing-app/blob/2feee8a3707d4cda887e3899597b8f85f601a38f/src/items/itembase.cpp#L2025). The reason I did not find the definition yesterday is, I used the github search function to find it - which is a bad idea [because it does not show all the matches in a file](https://github.com/fritzing/fritzing-app/search?utf8=%E2%9C%93&q=cachedConnectorItems).
Anyway, now it's clear that this function simply casts the content of the list returned by [`childItems()`](http://doc.qt.io/qt-4.8/qgraphicsitem.html#childItems) to the `connectorItem` type. Qt's documentation for `childItem()` says:
```The items are sorted by stacking order. This takes into account both the items' insertion order and their Z-values.```
I did not find out yet where the Items are added to the list. There are many functions altering z values, but I did not find out yet where the _relevant_ values come from.

htho, 2015-07-01:
As a workaround I have modified the Pushbutton. It is stretched to fit the 100mil grid.
https://github.com/htho/fritzing-parts/tree/master/core-pushbutton-stretched