fritzing/fritzing-app #3076: Raspberry v2 Part introduced in 0.9.2b is buggy

thommi2k, 2015-04-15:
* all 3v3 and GND pins in the GPIO header seem to be connected internally.
* unless GIPO is some new nomenclature I'm not aware of, all GPIOs are misspelled.

el-j, 2015-04-16:
thanks for the report.
- yes: gnd, 3v3 and 5v pins are connected internally, because they have the same signal. is that not correct? does it make any difference which gnd, 3v3 or 5v is used?
- the misspelling goes back to the first raspi version. thx and fixed

thommi2k, 2015-04-16:
Sorry, I did not phrase this well. What I mean is that all 3v3 pins are internally connected TO all gnd pins, surely that can not be right?

el-j, 2015-04-16:
hej again,
i'm sorry but i cannot confirm this bug. tested in all 3 view. the internal connections seems to be right from my point of view. 
[image]

thommi2k, 2015-04-16:
That does look right, yes. I will check when i come home.

thommi2k, 2015-04-16:
Well, there is some kind of bug, but it actually is not the Raspberry part...
Not sure how this can happen, I had a wire which shorted the 3v3 to GND as soon as one end was connected to the Raspberry 3v3, even if the other end wasn't connected to anything.

Screenshots:

[image]
(raspberry without the crazy wire: ok)

[image]
(raspberry with the wire just connected to 3v3: shorted to gnd ???)

el-j, 2015-04-17:
so, this is a bug with your sketch. as you see your green wire is connected to something, otherwise the end of the wire have to be "red" for "not connected". the green wire has green ending point. i do not know how that happen, but it's not a bug in the raspbi part.
so i close this issue

closed 2015-04-17 by el-j

thommi2k, 2015-04-17:
I'd say it's a bug in the wire part if it is possible to create such a wire, although I can't tell you how this happened.