fritzing/fritzing-app #3077: Problem with graphic design

TechKids, 2015-04-17:
Hello I have a problem with the new interface design. When I put an electronic component (LED, transistor etc) on the breadboard, there is no longer the highlight of the dotted line (like you have on your tutorials). There used to be green or yellow highlight and that was pretty handy. Also I can no longer connect a LED directly in the middle of the breadboard, it's possible to connect only on the sides of the breadboard. In the middle I don't have the blue highlight connectors. Am I missing something ? Thanks for your help, Alexandra

el-j, 2015-04-17:
hello alexandra. this seems odd. but i cannot confirm this behavior. what os do you use? what version of fritzing (32 or 64)? maybe you have some picture of the bug?
[image]

TechKids, 2015-04-17:
Hello,

Please find attached screenshots of what I get and what I want. As you can tell, the little dots don’t highlight. In the first place, I though it was because of the new version, so I rolled back to the old one. But it’s the same problem.

I’m using Mac OS 10.9.5 but also checked on Windows and it’s the same. It’s as if it does not recognize the connection ?! I can’t put the components into the dots!

Could you help please? Thank you,
- - - - - - - - - - - - - - - - -  
Alexandra & Tony
TECH KIDS ACADEMY

TechKids, 2015-04-17:
Hello again. OK so after several tests, I think I found the bug! Here’s what I do :

- Open Fritzing
- File / New
- The big breadboard displays
- I drag and drop a transistor or LED for example and it connects correctly, with the highlighted dots, that works well.

Now, since I want to use a half board, let’s do the process again (to explain the bug):
- Open Fritzing
- File / New
- The big / full breadboard displays
- Now I want to change the size of the breadboard and select Half+ in the drop down menu
- The half breadboard displays correctly
- I drag and drop components (transistor or LED for ex) onto the breadboard… the components don’t connect and the dots don’t highlight… that’s the bug!
- Now I move the breadboard on the screen….. and try to put back the components, here, now it works, they finally connect and the dots highlight!

That was a tricky one ;) I’m glad I found the problem & solution… maybe you could try to fix that in future release.

Thank you and have a nice day, 
- - - - - - - - - - - - - - - - -  
Alexandra & Tony
TECH KIDS ACADEMY

KjellMorgenstern, 2019-11-20:
I could reproduce a similar issue by disabling and enabling "align to grid"
1. disable align to grid
2. move breadboard in between the grid
3. enabling align to grid
4. new components will now be aligned to the grid, an might be unable to snap to the breadboard.

mMerlin, 2020-05-05:
I have seen the same (I think) issue. To reproduce (On Fedora 31, Fritzing 0.9.4):

1. open Frtizing app
1. breadboard view
1. verify that align to grid is on and 0.1" spacing
1. place resistor on breadboard (the default full+)
1. it aligns with and connects to the breadboard holes
1. move resistor off of the breadboard
1. select and change the breadboad to mini in Inspector
1. drag resistor over breadboard
1. it does not align with holes, and does not connect

similarly with the BB-301 breadboard, the resistor does not align well with the holes, but it does connect.

The issue seems to be that the different breadboard origins are not the same, relative to the grid that the connection points are on. When changing the breadboard size in the Inspector, the replacement breadboard is positioned at the same origin as the previous board.  That seems to be the upper left corner of the graphic. If the offset from there to the connection points is not the same as the previous breadboard, the connection points are no long on the grid, even though the original breadboard connection points were on the grid.

I have gotten into the habit of moving the replacement breadboard (or any part) as the first thing after changing it. That will align the connection points back to the defined grid.

A generic fix would be to have a way to assign the origin of the breadboard (or any part) to a specific location (in the part definition), instead of using the corner of the graphic. Having the origin match a specific connection point would prevent this sort of misalignment during substitution through Inspector. Pick the right connection point for the origin though. For example, the connection points on the outside rails of the half+ breadboard are not aligned with the .1" grid used for the main interior blocks of connections. I expect that the origin should be on the same grid as the centre blocks, not the rails.

failiz, 2021-03-19:
This big reports that this behaviour could also appear if you rotate 90 degrres the breadboard #2176

KjellMorgenstern, 2021-03-20:
It seems a solution would be to aling a part to the grid not only after move operations, but also after
* swapping with a different part
* rotating by 90 degree

failiz, 2021-03-20:
Could you change the title to be more representative of the problem?

failiz, 2022-01-22:
Align to the grid will work for rotated parts, but it is not enough for parts that have been swapped. 

After swapping, at least one of the connectors should be kept at the same position. As @mMerlin mentioned:
"A generic fix would be to have a way to assign the origin of the breadboard (or any part) to a specific location (in the part definition), instead of using the corner of the graphic. Having the origin match a specific connection point would prevent this sort of misalignment during substitution through Inspector."

Fritzing uses the connectors to align to the grid, and after swapping a part we should keep the first connector in the same place. This will solve issues when the old part and the new part have the same footprint for the connectors (for example, 3 pins with 100mils pitch), but the size of the image is different. This case can happen when fixing old graphics that have an unnecessary offset in the svg. See for example https://forum.fritzing.org/t/potentiometer-symbol/14790/1

In combination with issue #3927  (which will allow us to define which pin is the origin), it will minimize user interventions (remove wires that are added, move the part, etc.) when swapping parts.

KjellMorgenstern, 2023-09-30:
The rotation might look unexpected (and somewhat unpredictable) if we rotate around the alignment connector.
One option would be to align after rotating, but in that case the rotation would not be invertible: Rotating back would end up in a different position, so we dropped that idea.

Another idea would be to choose the grid point most near to the center of the part as rotation point. It think such a rotation would keep the part on the grid even if the center is not on the grid. And would be predictable, since the part still rotates mostly around the center.
To be invertible, we would have to remember the rotation point one it was chosen. 

In the long term, it might be possible to specify the alignment and rotation points directly in the SVG file, by extending the part format.