fritzing/fritzing-app #3078: Strange rectangle of copper when modifying a correct part (PCB)

martincote418, 2015-04-19:
Hello,
When I change pins size and position of core correct part, I get a strange rectangle of copper on both sides in PCB view.
The starting part SVG file is: sparkfun-connectors_power_jack_pth_pcb.svg
[image]

 And here the rusult:
[image]

The new SVG file is:
<?xml version="1.0" encoding="UTF-8" standalone="no"?>

<svg
   xmlns:dc="http://purl.org/dc/elements/1.1/"
   xmlns:cc="http://creativecommons.org/ns#"
   xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#"
   xmlns:svg="http://www.w3.org/2000/svg"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
   xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
   version="1.2"
   x="0in"
   y="0in"
   width="0.45120499in"
   height="0.51025999in"
   viewBox="0 0 3.2344361 3.6577694"
   id="svg2"
   inkscape:version="0.48+devel r13280"
   sodipodi:docname="sparkfun-connectors_power_jack_pth_pcb_1_35_mm.svg">
  <metadata
     id="metadata32">
    <rdf:RDF>
      <cc:Work
         rdf:about="">
        <dc:format>image/svg+xml</dc:format>
        <dc:type
           rdf:resource="http://purl.org/dc/dcmitype/StillImage" />
        <dc:title></dc:title>
      </cc:Work>
    </rdf:RDF>
  </metadata>
  <defs
     id="defs30" />
  <sodipodi:namedview
     pagecolor="#ffffff"
     bordercolor="#666666"
     borderopacity="1"
     objecttolerance="10"
     gridtolerance="10"
     guidetolerance="10"
     inkscape:pageopacity="0"
     inkscape:pageshadow="2"
     inkscape:window-width="1280"
     inkscape:window-height="738"
     id="namedview28"
     showgrid="false"
     inkscape:document-units="mm"
     showborder="false"
     inkscape:zoom="11.313708"
     inkscape:cx="23.022907"
     inkscape:cy="21.521268"
     inkscape:window-x="-8"
     inkscape:window-y="-8"
     inkscape:window-maximized="1"
     inkscape:current-layer="copper1" />
  <g
     id="copper1">
    <circle
       id="connector2pad"
       connectorname="PWR"
       cx="0.7055549"
       cy="0.52510363"
       r="0.21166661"
       style="fill:none;stroke:#f7bd13;stroke-width:0.14111108;stroke-miterlimit:4;stroke-dasharray:none" />
    <ellipse
       id="connector0pad"
       connectorname="GND"
       cx="0.7055549"
       cy="1.9644637"
       style="fill:none;stroke:#f7bd13;stroke-width:0.14111108;stroke-miterlimit:4;stroke-dasharray:none"
       rx="0.21166661"
       ry="0.21163952" />
    <circle
       id="connector1pad"
       connectorname="GNDBREAK"
       cx="1.4111103"
       cy="1.2588812"
       r="0.21166661"
       style="fill:none;stroke:#f7bd13;stroke-width:0.14111108;stroke-miterlimit:4;stroke-dasharray:none" />
    <g
       id="copper0">
      <ellipse
         id="circle9"
         connectorname="PWR"
         cx="0.7055549"
         cy="0.52510363"
         style="fill:none;stroke:#f7bd13;stroke-width:0.14111108;stroke-miterlimit:4;stroke-dasharray:none"
         rx="0.21166661"
         ry="0.21166663" />
      <circle
         id="circle11"
         connectorname="GND"
         cx="0.70555496"
         cy="1.9644367"
         r="0.21166661"
         style="fill:none;stroke:#f7bd13;stroke-width:0.14111108;stroke-miterlimit:4;stroke-dasharray:none" />
      <circle
         id="circle13"
         connectorname="GNDBREAK"
         cx="1.4111103"
         cy="1.2588812"
         r="0.21166661"
         style="fill:none;stroke:#f7bd13;stroke-width:0.14111108;stroke-miterlimit:4;stroke-dasharray:none" />
    </g>
  </g>
  <g
     id="silkscreen"
     transform="matrix(0.53893383,0,0,1.0109917,0.00565573,-0.05128228)">
    <line
       class="other"
       x1="2.5686736"
       y1="0.60931778"
       x2="1.976007"
       y2="0.60931778"
       id="line16"
       style="stroke:#f0f0f0;stroke-width:0.07769147;stroke-linecap:round" />
    <line
       class="other"
       x1="2.5686736"
       y1="3.6290958"
       x2="2.5686736"
       y2="2.1333179"
       id="line18"
       style="stroke:#f0f0f0;stroke-width:0.07769147;stroke-linecap:round" />
    <line
       class="other"
       x1="2.5686736"
       y1="0.60931778"
       x2="2.5686736"
       y2="0.80687332"
       id="line20"
       style="stroke:#f0f0f0;stroke-width:0.07769147;stroke-linecap:round" />
    <line
       class="other"
       x1="0.028673643"
       y1="3.6290958"
       x2="0.028673643"
       y2="0.60931778"
       id="line22"
       style="stroke:#f0f0f0;stroke-width:0.07769147;stroke-linecap:round" />
    <line
       class="other"
       x1="0.028673643"
       y1="0.60931778"
       x2="0.62134033"
       y2="0.60931778"
       id="line24"
       style="stroke:#f0f0f0;stroke-width:0.07769147;stroke-linecap:round" />
    <line
       class="other"
       x1="0.028673643"
       y1="3.6290958"
       x2="2.5686736"
       y2="3.6290958"
       id="line26"
       style="stroke:#f0f0f0;stroke-width:0.07769147;stroke-linecap:round" />
  </g>
</svg>

Thank for your help!

Martin

failiz, 2021-03-13:
Could you please upload the file? We cannot analyze the bug without it.  
Thanks!

KjellMorgenstern, 2021-03-14:
This looks like a problem with the IDs in the SVG file, probably they have not been preserved when editing it in inkscape.
As similar issue ( big rectangle connector in the center) is often visible in user created parts, and is due to missing connector ids.
Memory note: Next time we see this, create a ticket so we can add an automated check + warning

closed 2021-03-14 by KjellMorgenstern