fritzing/fritzing-app #3080: hole size and ring thickness cannot be set (french version)

el-j, 2015-04-29:
1. change the fritzing language to french
2. use a pinheader and try to change the values for ring thickness and hole size
3. it's not possible to write a "dot" (.) and fritzing does not take the new value

englisch and german version working quite well.
it should be tested if this behavior happens in other languages too

failiz, 2021-03-09:
Sill reproducible in 0.9.6
The problem comes from the line 957 and 970 in paletteitem.cpp (which is called by collectHoleSizeInfo() in the same file, which is called by collectExtraInfo() in pinheader.cpp). In those lines they create a QDoubleValidator for the field. 

From the QT documentation: "QDoubleValidator uses its locale() to interpret the number. For example, in the German locale, "1,234" will be accepted as the fractional number 1.234. In Arabic locales, QDoubleValidator will accept Arabic digits."

Thus, in french (and spanish, danish and other languages) the field only accepts "," instead of "." as decimal separator. But Fritzing does not convert the number after has been acepted. 

We should probably homogenize the use of the validators in Fritzing. For example, we have BoundedRegExpValidator in capacitor.cpp and resistor.cpp, but they a different pattern fro the RegExp.

Label bug, and maybe easy start?

failiz, 2021-03-09:
Related to #3140

failiz, 2021-03-12:
Mentioned in  #3140: "as done in calculator apps for instance, Fritzing should detect when the "," (main part of the keyboard) or "." (numpad) is pushed and interpret it automatically as a decimal point."

failiz, 2021-05-02:
I would set this to high priority. It can cause that users are not able to design their boards

failiz, 2022-06-19:
This also happens in the via as shown in #3956

blounolito, 2022-07-25:
Any update in this fix opened 3 months ago ?

KjellMorgenstern, 2022-07-25:
The issue state, labels and milestones mean the following:
Milestone: The fix will be part of the next (1.0.0) release. 
Label: The "solved" label means that we have a fix. 
Github state "open": In this case, the github issue will stay open until the fix is verified in a released version of Fritzing.

failiz, 2024-05-25:
It seems that it happens again in 1.0.2, see https://forum.fritzing.org/t/via-diameter-can-only-changed-with-integers/23700
I can reproduce it in 1.0.2 (Windows) when I select german as language.
Reopen issue.

failiz, 2024-05-28:
@KjellMorgenstern see above. Any idea of why the fix is not working?

KjellMorgenstern, 2024-06-02:
Yes, the input allows only "," , but the xml still allows only ".". Plan: Store the locale in the settings, only use it for UI elements. All XML elements should use the C locale. Make sure to transition the string when switching from Xml to Ui and back

failiz, 2024-06-02:
Nice that you found the issue, but lets re-open the issue

closed 2024-06-26 by KjellMorgenstern