fritzing/fritzing-app #3081: How does the panelizer works?

nraynaud, 2015-05-06:
Hi, I tried to register on the forum to ask, but I got a 500 when signing in. 

I'm overhauling the ground fill and gerber generation functions (to try to generate higher quality PCBs), but I see some code from the panelizer taping into the gerber.
I can't test it to check I didn't break it. it looks like it requires a specially crafted panelizer file. I see a python script that look like it generate the suitable file, but the documentation seems a little bit thin.

Do you have a good documentation on this thing ?

thanks,
Nico.

aknoerig, 2015-05-13:
Hi Nico, I'm afraid there is no good documentation on the panelizer. It is meant to panelize multiple sketches onto one larger rectangle for batch production, but it is a rather experimental feature. It's a multi-step process with intermediate config file generation.
@el-j will try out your pull-request on the panelizer, too, and if that causes problems we can go deeper into the panelizer documentation.

landracer, 2015-06-04:
I'm pretty sure we know what a panelizer is meant for. I'm pretty sure he was looking for instructions on using this feature. I would also like to see information regarding this feature. While I realize this software is in BETA development. If this feature exists, there should be a bit of documentation on this feature. Even if basic. This should not be some sort of top-secret feature. I'd like to use this feature myself. Now I am wasting countless hours screwing around, while I could be onto other things. I'm not far from leaving this EDA for something a bit more suitable for production situations. IE; gEDA, KiCAD. Nothing like starting from scratch on my design...

canadaduane, 2015-07-09:
I'm new here, but I think this is an open source project @landracer. In this type of project, the expectation is usually that the developers engage with the community, but that where something is lacking volunteers step up to fill in the gaps. For instance, if you've found yourself in a situation where an underdeveloped feature is a very important piece of your design cycle, you are uniquely qualified (motivated, aware of deficiency, etc.) to do something about it. If you feel you lack the skills necessary, you could communicate your willingness to work on this with others and form a smaller interest group with the skills to get the job done, and that would probably get the attention of core developers who are then more willing to shift their priorities from other endeavors to your group's interests.

closed 2019-07-18 by KjellMorgenstern