fritzing/fritzing-app #3084: Application freezes and crashes

spelare69, 2015-05-13:
When starting a new design and the steps below are followed the application freezes and stops responding. 

1. New design
2. breadbord tab is selected
3. delete default breadboard
4. Add single-sided stripboard by draging from core parts
5. Set size of stripeboard to columns = 39 and rows = 30
6. search for Component atmega328 and then drag Component to stripboard
7. Applicationa freezes and stops responding as soon as the Component touches the stripboard
As long as Component is not draged on top of the stripboard there is no problem.

This freezing is the same every time and I have tried 7-8 times.
No problem with smaller stripboard (default setting 30 columns x 20 rows) but when changing smaller stripboard design size to size 39 columns x 30 rows the application frezzes and stops responding.

Computer system:
windows 7 32-bit with service pack 1
i5 - 4670 3.40 GHz, 4GB of RAM

BR
Spelare69

aknoerig, 2015-05-13:
I'm afraid that's due to the relatively naive implementation we made. It's probably not freezing, but just taking a lot of time to compute the stripboard behaviour. More optimization work is needed here.

Gemba, 2015-06-09:
This behavior is caused by #3090 (see there for a workaround). It is unrelated to the selected component. Verified with 0.9.2b on Linux.

thkfighter, 2018-10-26:
Fritzing is freezing. It is fritzing 0.9.3b 64bit on Windows 10 Home Edition (upgraded to the newest on Oct 26, 2018). Each time I start fritzing, the application promts "not responding" or just freezes without any prompt.
When I run fritzing as the administrator, it does not freeze at the start but do after I do some operations like the ones mentioned by @spelare69 .

PC    Lenovo 20KFA01PCD laptop
OS    Windows 10 64bit home	
CPU  Intel Core i5-8250U @ 1.60GHz
Motherboard    Lenovo 20KFA01PCD ( Intel Xeon E3 - 1200 v6/7th Gen Intel Core/DRAM Registers - 5914 - 7th Generation Intel...)
RAM    8GB
HD    Sandisk SD9TN8W256G1001 ( 256 GB / SSD )
GPU  UHD Graphics 620 ( 128 MB )
Network card    Intel Ethernet Connection  I219-V

KjellMorgenstern, 2019-12-04:
Gemba is right: Because of #3090 all connectors of the board are connected to each other. This is not a typical case for a stripboard, and should be avoided. Furthermore, when Fritzing tries to build a list of connected items, it uses said naive implementation.

closed 2021-09-24 by KjellMorgenstern