fritzing/fritzing-app #3085: Headers don't reflect settings

holgerlembke, 2015-05-15:
- create new schematic
- add generic female Header
- set 8 Pins

Change view to pcb view

- set form to long pad
- set pin distance to 2 mm

--> settings are accepted, but layount does not change

marc-gist, 2015-10-30:
This is also true when changing the "row" from single, to double.  It still displays as a single row on the Breadboard, but the PCB shows the double row.

chrisyokum, 2015-12-18:
This seems to be an issue regardless of whether you try to place a "Generic Female Header" or "Generic Shrouded Header" - both will render nicely on the PCB but are unusable on the Breadboard/Perfboard.

Is there some way I can fix this myself?

chrisyokum, 2015-12-18:
Nevermind.. "Mystery part" to the rescue!

chrisyokum, 2015-12-18:
Whenever this is fixed, it would be nice to use the pin numbering in zig-zag form. I notice that the "male shrouded" has the numbers counter-clockwise/anticlockwise from 1. It should be zig-zag across (odd on one side, even on the other) for these expansion connectors.. https://en.wikipedia.org/wiki/Insulation-displacement_connector

failiz, 2021-03-13:
Confirmed in 0.9.6
Add bug and PCBView labels

failiz, 2021-03-13:
Related to #2199 #2923 and  #2758

KjellMorgenstern, 2021-12-09:
Closing in favor of #2625

closed 2021-12-09 by KjellMorgenstern