fritzing/fritzing-app #3093: mousewheel with hover on view mode buttons plays roulette

designerfuzzi, 2015-06-07:
when you hover with a mouse over those upper view mode buttons, touch your mousewheel and look what happens.the app is iterating thru all view modes until ist stops at a random mode of those 5 views.
tried to solve that by turn mousewheel off, but stil plays this las vegas roulette with them.
point is, i could not find where the coins fall out of my computer.

tylerhunt, 2015-07-09:
In general, I'd say the mouse wheel is far overused as an interface navigation mechanism in the app. I inadvertently trigger some kind of scrolling or context switching much more often than not. This seems to be made worse by the sensitivity of the scrolling, especially when using an Apple Magic Mouse.

htho, 2015-07-10:
@designerfuzzi I don't have this issue. (Version 0.9.2b on Linux) You meant using the scroll wheel on the top row right below the menu? Scrolling stops either on code, or on welcome.
@tylerhunt Hmm I did not have that feeling yet. Where can you give an example?

tylerhunt, 2015-07-10:
@htho For instance, scrolling when hovering over the groups in the parts selector doesn't actually scroll the list, but instead it switches between the groups. This behavior is unexpected, and especially annoying due to the load time required for each group initially.

The expected behavior would be to have scrolling move up or down in the list without actually changing the selected group. A group is selected by click on it.

htho, 2015-07-10:
@tylerhunt right after I read this issue, I opened an [Issue addressing that problem](https://github.com/fritzing/fritzing-app/issues/3104). I would not call *that* overusing.

tylerhunt, 2015-07-10:
@htho I'd argue that using the mouse wheel to select the tabs at the top is wrong, too. I can't recall ever using another app that did something like that. Scrolling with the mouse should always just scroll and never actually change a selection.

designerfuzzi, 2015-07-10:
i agree with @tylerhunt , the expected behaviour in a desktop app is that the main navigation is not sensitive to mouse wheel and instead action take place when clicked only.
It is also unexpected that both "wheel-directions" of a standard magic mouse scroll thru all of the 5 groups until it stops at a random group, depending on the amount of non-precise clicking on the mouse (which triggers wheel action). The actual behaviour tells that the mouse events still bubble on the window after they triggert the change of group action.

htho, 2015-07-10:
I just tried some programs (on Linux)
* Geany and PcmanFM (gtk+): change tabs
* qjackctl (Qt): change tabs
* firfox and thunderbird: scroll if there are so many tabs that scrolling is necessary. They do nothing otherwise.
* chromium: change tabs
* Eclipse does nothing.
Luckily those top-rows of tabs we remember from windows configuration Dialogs are almost dead. A pity Fritzings configuration dialog still uses them in the *worst* way (horizontal scrolling). Other programs have a list widget on the left, which is waaaay better and behaves as you and I would expect it.
Maybe the part list was tabbed once, in fact it is a list now, so it should behave like one.

I still think the change behaviour is appropriate for tabs. It is not for lists.

designerfuzzi, 2015-07-10:
Just to remind: Mousewheel is a vector based input, it tells more about direction of the vector the user choose than it tells about amount of fields to climb up the vector. The idea of mousewheel is missused in many forms. Like coded actions no mater which direction was selected/trigger from the user. So this kind of input should be used only if a vector based navigation is needed. In modern view programming this is only the case if the end of nodes on a list (vector) is more than one visible pagination away.

htho, 2015-07-10:
@designerfuzzi I have never seen it like this, but you are absolutely right: Scrolling should not be expected to give discrete values. Anyway, computers just know discrete values, and most desktop mice actually give haptic feedback on discrete values.

Right now I am using a mouse with a discrete wheel: Changing tabs using this mouse seems reasonable. I'd never change tabs using my touchpad though. - Interesting

tylerhunt, 2015-07-10:
@htho It sounds like some of this might be chalked up to differences in OS conventions. Maybe this is used more often in Linux. (Just checked in Chrome on the Mac and scrolling in the tab bar does _not_ switch tabs.)

Regardless, I second @designerfuzzi's assessment.

htho, 2015-07-10:
@tylerhunt You probably are right.
@designerfuzzi You said this "as a reminder". I have never heard of it. Where do I get the book?

designerfuzzi, 2015-07-11:
^^ some years software ergonomics consultant (GUI and HID, head of industrial design in manufacturing) in automotive industry with 12 brands where thousands of processes are controlled from hundrets of engineers while millions of parts in different product groups and manufactury chains on different continents have to be organized just in time and all of those can change every day. So tiny changes can have huge effect in logic while helping people focusing on the product. The most expensive part in development is not the product, it is the human resource who needs time to get around problems which come not with the product, instead come with the workflow.
Maybe i should start a kickstarter for your book. :) A last word on tabs, tabs are mostly a design pattern which is representing different files. Going back where that patterns meta language comes from, flick through all open files (even if cpu provides quick access to files on the fly) is an expression of search. Once you got the right file, you pull it out. You would not pull out all files in search of just one, excepting in case of a very unsorted files base. So tabs in webbrowsers are expected to load or refresh the data after the file was requested. In Fritzing this tabs represent different parts of the workflow not different files or pages.

closed 2019-05-03 by KjellMorgenstern