fritzing/fritzing-app #3095: How do I create arbitrary junctions for copper paths in PCB view?

Zalewa, 2015-06-10:
I want to create a lane for my +5V signal that would branch into various elements. I'd like the lane to go through the board and branch off to the elements where the path is the shortest, like this:

[image]

However, I can't figure out what is the proper way to branch off copper paths like this. I can connect the elements directly and bend the paths, but then Fritzing sees these paths as different objects that occupy the same space. When I want to move the path or it's corner, I have to move every single one of them.

htho, 2015-06-20:
The "solution" is to make a connection to the last element and see it as the "lane". Then add bendpoints for each element and connect them to their bendpoints. Its not beautiful, but it works.

Zalewa, 2015-06-20:
I guess it's a solution, albeit a cumbersome one when compared to how Eagle operates in this department.

Elijahg, 2015-06-22:
You can hold option/alt and drag from a bendpoint (which you can double click to create).

failiz, 2021-03-19:
Duplicate of  #1386

KjellMorgenstern, 2021-03-20:
https://user-images.githubusercontent.com/6494025/111868102-44e6f680-8978-11eb-995f-e57dd1ee8520.mp4

closed 2021-03-20 by KjellMorgenstern