fritzing/fritzing-app #3096: Part editor: Error: "there are no copper layers defined in imported SVG-File"

leunert, 2015-06-14:
I tried to modify a certain part from 6 Pin-Version to 5 Pin version. So I exported the PCB file, Deleted one Pin with inkspace and imported the file again. Then I get above mentioned error. Even when i reimport the file without any changes I get the same error. What do I have to do. Is there anyone who has a solution for this issue ?

Elijahg, 2015-06-17:
Can you post the SVG file?

failiz, 2021-06-23:
There is no file, so we cannot debug this. Close?

closed 2021-06-24 by KjellMorgenstern

bugith, 2024-11-17:
Here is one that gives this error
[image]

KjellMorgenstern, 2024-11-17:
@bugith All the IDs are just some generated numbers, like "g135732". This is quite a broken SVG. It has been processed with Adobe and Inkscape at least. Not only all the IDs are lost, also the scale is lost (it is in pixels, without any way to determine the DPI, so all are relative sizes). Don't use pixel, use mm or inch for the SVG width and height. 
Using tspan also is not fully supported (it is a svg helper to format text to form a circle or similar operations), using CSS styles is not supported, use svg xml attributes instead. Font sizes must be relative to the document, absolute font sizes in pixel don't make any sense for documents that need to be in scale, so you can't use fonts size in "px"

I think you need to start from scratch with a new file. Check the forum and fritzing.org on tips and tricks how to create Fritzing Parts.

We are adding some improvements in error reporting for Fritzing.