fritzing/fritzing-app #3100: create Uninstall-File

el-j, 2015-06-29:
it would be good to have a uninstall file to clean up the user-folders.

failiz, 2021-03-12:
This is a valid issue, but maybe we should create a label for "install/uninstall" issues

KjellMorgenstern, 2021-03-12:
I don't really see deleting user folders as a valid feature? The uninstaller only removes the application, not the data. This works fine for Windows and Mac with version 0.9.6
Registry data and user folders are not deleted, but this is intended. 

Close the issue?

failiz, 2021-03-12:
I think sometimes they ask if you want to remove the data too. Another thing would be to make an uninstaller, but not sure if the app writes anything in the register or if it just the folder.
In any case, it does not seem a very demanded feature and there plenty of other urgent features waiting for us. So, closing the issue is fine with me.

closed 2021-03-12 by KjellMorgenstern