fritzing/fritzing-app #3105: Wiring perfboard

tylerhunt, 2015-07-10:
I'm finding it very difficult to use the perfboard. Unlike the stripboard, which˙has the ability to cut and build traces, it appears the only way to connect things on the perfboard is to use wires. The issue I'm having, though, is that if I'm trying to wire two components together on the perfboard, it looks like they're connected, but it doesn't appear there's actually a connection made.

How are you supposed to connect components on the perfboard? Am I missing something?

htho, 2015-07-10:
You can do this with the striped perfboard by clicking between the holes.

tylerhunt, 2015-07-10:
Right, but if I'm not using a stripboard, that would require me to go through and cut all the traces to make it look like a regular perfboard. Surely there should be a better way to do this. Maybe the default perfboard component should behave like the stripboard and traces to be created.

Why don't wires seem to work properly on perfboards?

htho, 2015-07-10:
That is why I avoid this view and use the PCB view instead.

tylerhunt, 2015-07-10:
@htho Again, you're providing workarounds without addressing the actual issue.

htho, 2015-07-10:
I'd fix it if I was a developer. I am just a user coping with the same issues and sharing my experience with you.

tylerhunt, 2015-07-10:
@htho Gotcha. Thanks for the suggestions.

otede, 2015-08-15:
Perfboard seems to be really porked in Fritzing. I constantly get problems with it. See https://github.com/fritzing/fritzing-app/issues/3130

aknoerig, 2015-08-16:
The perfboard and stripboard aren't really first-class citizens in Fritzing. We added them relatively late in the process, because with the people we talked to everyone was using breadboards.
As you all are showing, perfboards require a different set of interactions, specifically a front/back view and better tools and part designs for the very dense wiring. 
This is no dead end though and could be optimized for the future of course, just not a high priority for us, so any help is welcome

otede, 2015-08-16:
@aknoerig That's fine, but even with single-side usage there are frequent quirks. Also, a prior notice on the actual state of the said features would be more than welcome.

core2zero, 2015-11-30:
it's impossible to make a perfboard with a mix of vertical and horizontal traces, half an hour i'm still trying to make the third strip

KjellMorgenstern, 2019-11-09:
Trying to reproduce the original issue "Wire components together".
It seems to work fine. I tried with a wire and by connecting them on the same hole. Connections were made, and also showed up in schematics and pcb view. Same for stripboard.

While there are some improvements possible, like front / back view and connecting with solder instead of wires, I could not reproduce any problem with connecting or wiring up components.

closed 2019-11-09 by KjellMorgenstern