fritzing/fritzing-app #3108: IC rendering glitch in schematic view

tylerhunt, 2015-07-10:
I've added several 4-pin ICs to a schematic, and everything looked fine, but when I closed and reopened the app, it no looks like this:
 
[image]
 
The bounds of the IC are kind of duplicated and shifted to the right, and the pin labels are no longer properly aligned. This IC was rotated 180 degrees, in case that has something to do with it.

This is what those ICs should look like:

[image]

KjellMorgenstern, 2019-05-03:
This issue was closed because it is outdated.
If you think this issue is still important, especially if version 0.9.3b still has this issue, please open.

closed 2019-05-03 by KjellMorgenstern

oxygene, 2020-01-27:
I just started using Fritzing 0.9.4, created a generic IC with 16 pins, rotated it in schematic view by 90 degrees. Then I saved and reopened the project, and the pin labels seems to have shifted to the bottom right. Seems it keeps shifting on subsequent save/reload cycles.

[image]