fritzing/fritzing-app #3109: Parts can't be reordered

tylerhunt, 2015-07-10:
Sometimes the parts get out of order, like how the Spark Core is beneath the headers here:

[image]

Selecting "Bring to Front" doesn't do anything in this case.

I've also seen parts end up behind the stripboard and been unable to "Send to Back" the stripboard.

mobilinkd, 2015-07-11:
I was about to report the same bug.  I can confirm that this exists in the Linux version of the product (0.9.2b)

Artyrm, 2016-09-14:
I confirm this in Win 0.9.3

beerygaz, 2017-09-25:
Confirmed on 0.9.3.b Win 64 too.

failiz, 2021-11-02:
I think that this is a duplicate of #3627, which has been solved