fritzing/fritzing-app #3115: Bend points are broken after save / crash-restore

thijs, 2015-07-13:
[image]

Today I've twice returned to a project to find all the bend points in my traces broken. See the screenshot above. The problem is not just visual, see below:

[image]

When I try to move the bend points the traces seem to have been doubled up multiple times on each other and everything gets even more screwy. So the only option is to delete all traces and start over.

I've now seen this twice, as I said. Once after choosing to restore after a crash, once after just a regular save and then opening the project again later in the day.

ubenedix, 2015-08-05:
Same problem.
Reopend files are scrap. 
Moving bend points leads to chaos. 
1: file reopend -> broken connections everywhere 
2: moved bend point at broken connection: bend point not common to 2 traces 
3: moved bend point at "C5" just a little bit to left -> "even more screwy":  
I only had to touch 2 point to get from 1 to the chaos in 3. 
This makes frizzing 0.9.2 ... well, unhelpful. 

[image]
[image]
[image]

OnkelDittmeyer, 2015-10-06:
Same issues with Fritzing 0.9.2 under OS X10.9.5

gepd, 2015-12-01:
I'll let here this gif to undertstand better this problem.

[image]

aknoerig, 2016-01-24:
A quick heads up to everyone experiencing this bug: Jonathan has made great progress on it in the "crossed" branch (https://github.com/fritzing/fritzing-app/tree/crossed). The bug will be fixed from 0.9.3 on, and will also fix 99% of all broken files created with earlier versions.
Thanks for all the insights you provided!

gepd, 2016-01-25:
Excellent news, I've been waiting long for it! :)

aknoerig, 2016-01-28:
Fixed and pushed in 12b7894

closed 2016-01-28 by aknoerig

crazymadprofessor, 2016-02-06:
Same issue using Maveriks, 
Yesterday I been doing a PCB project for 7 hours non-stop, Today morning when I opened my project I realized that my project was usefulness , and  if I try to rectify the wire routing, my project get chaotic.

Is it there any solution? 
Is there some way I can fix that problemt?

Thank you very much

aknoerig, 2016-02-14:
Sorry that this nasty bug has hit you, too. Expect a release of the new fritzing version in the next few weeks. It has the bug fixed, and will also repair old broken files. 

If you don't want to wait, you can try building the latest version yourself, but that's a few hours work: https://github.com/fritzing/fritzing-app/wiki/1.-Building-Fritzing

crazymadprofessor, 2016-03-05:
Hey thank you very much ,  just I read your mail . 

When my project on fritzing crashed I stopped for a while, and experimented with eagle and other softwares, but to be honest are not the same than fritzing, the workflow in fritzing is much faster and much more intuitive than other freewares . you done a really good job . Actually I don't have to much money , but definitely if I could start to work with fritzing again I will try donate some to this fantastic project.

Thank you so much for all the effort.

Regards

Date: Sun, 14 Feb 2016 11:37:21 -0800

mearsy, 2016-03-07:
Any idea when 0.93 will be on the main site for download?

aknoerig, 2016-03-07:
We're still refining a few things while we're waiting for the final release of Qt 5.6 (which brings High-DPI support). It's scheduled for sometime in March, expect our release shortly after.

mearsy, 2016-03-30:
Now qt 5.6 is out any idea when fritzing 0.93 will be released. im kind of stuck until i can fix my bendpoint issue

aknoerig, 2016-04-03:
Getting really close now.. Note that Fab is already using it, so pcb orders will be fine.
Which OS are you on? Maybe we can issue a pre-release.

mearsy, 2016-04-04:
Osx