fritzing/fritzing-app #3117: Keyboard Shortcuts issue

Hobbitinn, 2015-07-24:
I'm running Ubuntu 14.04 and the amazing Fritzing 0.9.2.

I'm having a really annoying problem with Keyboard Shortcuts.
is there any solution to this, is this a known problem?

cheers! and thanks for all your hard work! :D 
Arni Geir

htho, 2015-07-24:
And what *is* the problem?

(btw. I am not a Fritzing-Developer)

Hobbitinn, 2015-07-25:
The problem is that none of the keyboard short cuts work.
for example: 

Ctrl+0 won't fit the schematic to window.
Ctrl++ won't zoom in
Ctrl+- won't zoom out
Ctrl+z won't undo... etc. etc.

This slows down the work flow quite a bit, as you could imagine.

Thank you :)

Kent42, 2015-12-24:
I am running Ubuntu 14.04 and upgraded from .8.7 to .9.2 the keyboard controls do not work any longer. Even the delete key. Everything has to be done via the mouse.

aknoerig, 2016-03-20:
Can someone try this with a recent build based on Qt 5.5+? This is likely a Qt issue.

mmatl, 2016-03-21:
I'm experiencing the same issue.

aknoerig, 2016-03-22:
Then it might be related to this workaround we used for adding "Code View":
https://github.com/fritzing/fritzing-app/blob/master/src/program/programwindow.cpp#L183

What is now Code View use to be called ProgramWindow, and was a opened in a separate QMainWindow, with its own shortcut bindings. It still is that architecture, but squeezed into the Code Tab.
The workaround employed here for the shortcut keys works fine on Windows and OSX. It could be that it's now working on Linux with the latest Qt, too. It would be great if someone tried this and also give the  function above a closer look.

vajpeters, 2016-05-03:
Just checked it by running it through Qt 5.6 and this seems to solve the issue...

aknoerig, 2016-05-04:
Thanks, also works for me.

closed 2016-05-04 by aknoerig