fritzing/fritzing-app #3120: pcb wire trace issue

syedubaid, 2015-07-25:
when selecting large wire in pcb it covers the hole for "through hole" components

bootchk, 2015-08-08:
Is this really a problem in the actual board?  In other words, is it just a user-interface issue, that it would be nice if Fritzing drew it better?

In other words, when the fab creates the part, won't they drill and plate the hole, no matter what the top copper layer says?  They etch the copper layer, then drill, then plate the holes.

aknoerig, 2015-08-10:
It's definitely not a production issue, only a rendering issue. The problem also exists in the other direction, see https://github.com/fritzing/fritzing-app/issues/3128. 
The maths for this is a little tricky because one needs to take into account quite a few things: hole diameter, ring thickness, trace thickness (which defines the size of the rounded cap).

cadop, 2016-01-07:
This issue came up for me, and causes a problem when exporting the svg, since the wire overlaps the hole, the traces make the hole not exactly a circle, and the pcb manufacturing can not locate what the center of the hole is.

bootchk, 2016-01-07:
cadop: you might try a different output format such as Gerber (which might have a separate drill hole file giving the centers of the holes.)  If your fab shop can use Gerber.

cadop, 2016-01-08:
So far, the best solution I could find was decreasing the hole size and increasing the pad size.  Its just not efficient when dealing with premade components, but works well for using headers.

KjellMorgenstern, 2019-11-09:
This issue got worse with #3379 . Probably we will have to cut a circle from the trace cap on through holes and vias (and draw over a circle for formats like svg)

KjellMorgenstern, 2019-11-09:
As already stated by @bootchk, the drill file from gerber export can be used as a workaround.

failiz, 2021-04-27:
As stated in #2653, the same problem appears with curved tracks: "If you draw a curved trace, the ends cover the holes they extend from."

mjr129, 2024-06-12:
I see this problem in production too (*_etch_copper_bottom/_mirror). This makes the hole off-centre which can cause the drill to slip into the wrong position.

KjellMorgenstern, 2024-06-12:
Testfile with various combinations of hole sizes and wires. 
For straight wires, some offset is already applied to mitigate the issue, but a proper solution would just substract the hole from the wire.

[issue3120.zip](https://github.com/user-attachments/files/15804058/issue3120.zip)
[image]

KjellMorgenstern, 2024-06-12:
Using the gerber export, the silk, mask and copper layer  (without the drill file)
[image]

With the drill file:
[image]

KjellMorgenstern, 2024-06-12:
Fritzing doesn't try to etch the copper on top of holes. Depending on the process, one might prefer to etch the copper before drilling, or just drill through the copper. 
@mjr129 I guess you are using SVG or PDF export? How do you determine the hole position, is the drill reflected from asymetric copper leftovers?

mjr129, 2024-06-12:
@KjellMorgenstern Yes, should have said, that's the PDF export:

[image]

And this only affects drilling the holes manually. Actual hole positions are fine:

[image]