fritzing/fritzing-app #3128: PCB traces not intersect tiny SMD pads

bootchk, 2015-08-08:
On PCB view, copper traces do not appear to intersect tiny SMD pads.

To replicate, add a wire connecting a via to a mosfet in the SOT323 package (from SparkFun Discrete bin) and reduce the wire width to extra thin (12 mil).  Image attached.  I can provide a small sketch.

I am not sure this is really a problem for an actual board.  It appears that generated Gerber might partially intersect.  Users should check their Gerber copper layers.

It certainly is not a problem for all users, only those who are attempting to use very small SMD packages say SOT323 and possibly packages smaller than 603 (metric 1608) which are so small that parts are difficult to place by hand, and usually require reflow soldering.

A workaround is to add a bendpoint to the wire so that the wire passes through the pad, reverses course, and intersects on the far side of the pad.  This workaround does leave a little, harmless bulge of copper.

[image]

bootchk, 2015-08-10:
aknoerig:  I agree it is not a serious issue.  However, it is flaky.  When I opened the same sketch today, it looks different.  Maybe it depends on the zoom level.

More importantly, the Gerbers are probably OK, but in some views (speficially thumbnails and previews) they look suspicious.  When I submitted my sketch to OshPark fab, the view their website presented looked suspicious enough that I implemented the workaround.

Here are the Gerber views using an online Gerber viewer.  It looks OK in the main view (but the trace is pinched a little where it meets the pad.)  But the thumbnail view in the lower right looks worse.  I know this is depends on the resolution and the rendering algorithms.

I suppose it is a tradeoff: if you rendered the end of the trace to the center of the pad, then you would get more complaints about the hole not being there.

And I suppose you don't want to add special code to insure that the circle where the end of the trace intersects the pad is smaller than the pad itself. 
[image]

aknoerig, 2015-08-10:
Yep, as written in issue #3120, this calls for refinement of the path length calculation.

KjellMorgenstern, 2019-11-09:
Issues is fixed.

closed 2019-11-09 by KjellMorgenstern