fritzing/fritzing-app #3129: 0.9.2 crashes on Mac OSX 10.7.5

3dave, 2015-08-10:
Hello,

Newbie here - sorry for eventual repost. Fritzing repeatedly crashes on various actions: whether dragging new components or doing an edit like rotating etc. It also does not happen all the time so it is completely random.

I have several crash-reports, this is the last one. I hope it can help find the bug.

Cheers,
G

==============================================================================

Process:         Fritzing [1129]
Path:            /Applications/Fritzing.app/Contents/MacOS/Fritzing
Identifier:      org.fritzing.Fritzing
Version:         0.9.2 (bda4e)
Code Type:       X86-64 (Native)
Parent Process:  launchd [170]

Date/Time:       2015-08-10 17:31:02.032 +0200
OS Version:      Mac OS X 10.7.5 (11G63)
Report Version:  9

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: KERN_INVALID_ADDRESS at 0x0000000011b69f1d

VM Regions Near 0x11b69f1d:
--> 
    __TEXT                 0000000100000000-0000000100960000 [ 9600K] r-x/rwx SM=COW  /Applications/Fritzing.app/Contents/MacOS/Fritzing

Application Specific Information:
objc[1129]: garbage collection is OFF

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   QtWidgets                     	0x0000000100bf80a1 QWidget::mapFromGlobal(QPoint const&) const + 49
1   QtWidgets                     	0x0000000100bfcbc6 0x100b79000 + 539590
2   QtWidgets                     	0x0000000100bfa4a1 0x100b79000 + 529569
3   QtWidgets                     	0x0000000100ba4f2c QApplicationPrivate::notify_helper(QObject*, QEvent*) + 300
4   QtWidgets                     	0x0000000100ba7a1d QApplication::notify(QObject*, QEvent*) + 6157
5   org.fritzing.Fritzing         	0x000000010003bedf FApplication::notify(QObject*, QEvent*) + 31
6   QtCore                        	0x00000001017a60b2 QCoreApplication::notifyInternal(QObject*, QEvent*) + 114
7   QtGui                         	0x000000010117885f QGuiApplicationPrivate::processTabletEvent(QWindowSystemInterfacePrivate::TabletEvent*) + 911
8   QtGui                         	0x0000000101174630 QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) + 944
9   QtGui                         	0x0000000101164338 QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 136
10  libqcocoa.dylib               	0x000000010451a4c7 0x104500000 + 107719
11  libqcocoa.dylib               	0x000000010451afd8 0x104500000 + 110552
12  com.apple.CoreFoundation      	0x00007fff8e0394f1 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
13  com.apple.CoreFoundation      	0x00007fff8e038d5d __CFRunLoopDoSources0 + 253
14  com.apple.CoreFoundation      	0x00007fff8e05fb49 __CFRunLoopRun + 905
15  com.apple.CoreFoundation      	0x00007fff8e05f486 CFRunLoopRunSpecific + 230
16  com.apple.HIToolbox           	0x00007fff8e3092bf RunCurrentEventLoopInMode + 277
17  com.apple.HIToolbox           	0x00007fff8e3104bf ReceiveNextEventCommon + 181
18  com.apple.HIToolbox           	0x00007fff8e3103fa BlockUntilNextEventMatchingListInMode + 62
19  com.apple.AppKit              	0x00007fff8f0b8779 _DPSNextEvent + 659
20  com.apple.AppKit              	0x00007fff8f0b807d -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 135
21  com.apple.AppKit              	0x00007fff8f0b49b9 -[NSApplication run] + 470
22  libqcocoa.dylib               	0x0000000104519d44 0x104500000 + 105796
23  QtCore                        	0x00000001017a364d QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 397
24  QtCore                        	0x00000001017a6652 QCoreApplication::exec() + 354
25  org.fritzing.Fritzing         	0x000000010004ff11 main + 817
26  org.fritzing.Fritzing         	0x0000000100005ed4 start + 52

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff9286e7e6 kevent + 10
1   libdispatch.dylib             	0x00007fff91c0e786 _dispatch_mgr_invoke + 923
2   libdispatch.dylib             	0x00007fff91c0d316 _dispatch_mgr_thread + 54

Thread 2:: Qt bearer thread
0   libsystem_kernel.dylib        	0x00007fff9286ddf2 __select + 10
[... 239 lines cut ...]
CG shared images                   3440K
CoreAnimation                         4K
CoreGraphics                         16K
CoreImage                             8K
CoreServices                       1800K
MALLOC                            378.2M
MALLOC guard page                    48K
Memory tag=240                        4K
Memory tag=242                       12K
OpenCL                               28K
STACK GUARD                        56.0M
Stack                              9756K
Stack (reserved)                    520K        reserved VM address space (unallocated)
VM_ALLOCATE                        16.2M
__CI_BITMAP                          80K
__DATA                             13.3M
__IMAGE                             528K
__LINKEDIT                         53.5M
__TEXT                            127.5M
__UNICODE                           544K
mapped file                       132.7M
shared memory                       312K
===========                      =======
TOTAL                             867.7M
TOTAL, minus reserved VM space    867.2M

aknoerig, 2015-08-10:
This seems to be related to the use of a graphics tablet, compare http://stackoverflow.com/questions/21735462/qt-app-crashes-on-every-message-box
This will likely disappear with an upgrade to a newer Qt version. For now, try disconnecting the tablet.

3dave, 2015-08-11:
Wow, right on the spot. I do have a Wacom Intuos M, all with the latest driver. I worked a couple hours with the tablet disconnected - using a mouse and it does stop crashing!

Can you pls be more specific on how I can upgrade to a newer Qt - as far as I know I didn't  install it myself, does it come with Fritzing?

Thnx!

aknoerig, 2015-08-12:
:smile:
Upgrading Qt can only be done by developers. It's the underlying C++ framework we're using. We're at 5.2 wheres the latest is I think 5.5. It's always risky to upgrade, though, so it might be a while until someone gets around to it.

KjellMorgenstern, 2019-11-20:
We are on >= 5.9 now (5.12 on some platforms), so I think this issue is solved.

closed 2019-11-20 by KjellMorgenstern