fritzing/fritzing-app #3130: Unwanted connections created on perfboard

otede, 2015-08-15:
(on perfboard because I haven't noticed anything like this on breadboard)

See the files first:
http://1drv.ms/1DIvJg7

CASE 1
Initial state:
Adding a screw terminal J1pan-board to the circuit has caused an unwanted connection to be created (R1, pins 1 and 3).

Steps from there:
see the blad.png file
 
Files:
bug_terminal_unwanted01.fzz
bug_terminal_unwanted01endstate.fzz
 

CASE 2
Initial state:
No unwanted connections. See file bug_terminal_nounwantedconns.fzz
 
Steps from there:
Screw terminal Terminal J1pan-board moved - unwanted wire3147 gets created. See file bug_terminal_unwanted_wire3147.fzz

CASE 3
Another simple example. Multiple phantom connections.

An additional note:
If I hover over a phantom connection I get its name. Then, when I go to Breadboard view and enter the name in Part > Find part in sketch... it returns: "No parts matched search term '<phantomwirename>'.
[image]
[image]

Fritzing Version 0.9.2 (b8d2d5970658f0bed09c661c9ea9a515b5f40f44c 2014-11-19) 64 [Qt 5.2.1]
Windows 7 x64

phamiltonsmith, 2015-10-02:
Any news on this? I am having the same issues... For some reason the app is creating connections that do not exist...

KjellMorgenstern, 2019-11-20:
The files on OneDrive are no longer available. 
Is the issue still reproducable?
If yes, please attach files to this issue.

otede, 2020-01-12:
> 
> 
> The files on OneDrive are no longer available.
> Is the issue still reproducable?
> If yes, please attach files to this issue.

I don't have time for this anymore. Here are the files on onedrive
https://1drv.ms/u/s!AvyUQyNGJs9mhPFmbM16Q43lZ7W2IA?e=fGFfZo