fritzing/fritzing-app #3131: Is Fritizing actively developed

otede, 2015-08-15:
Are there any official announcements on plans regarding Fritzing?
Im in the process of choosing a set of tools for my initial dive into hobby electronics and I think it's a fair thing to do to let people know whether a certain project has a future so they don't waste their time.

el-j, 2015-08-16:
yeah it is.

otede, 2015-08-16:
But with no official announcements on plans regarding Fritzing?

duff2013, 2015-09-18:
The state of the website, spam filled forum and no activaty on here for months i would say its pretty much dead. To bad its was such a great peice of software but I guess I'll just have to use Altiums CircuitMaker for now on.

el-j, 2015-09-19:
hje hej, i know it seems like dead. but it isn't. we need to organize really much in the background. e.g. the forum we use will be replaced by a new one in the next days. sadly the spam-bots are so hard on the old one that we are not able to delete the spam faster than they create a new entry. altiums seems nice. but fritzing is not dead! ;)

duff2013, 2015-09-19:
Great!!! Altium is nice but no OSX support and still has a learning curve where fritzing is easy. I hope in the future you can implement a proper measurement system when designing PCB's. That ruler with 1 cm resolution makes it difficult to size things correctly but anyway glad to here things are still in development!

otede, 2015-10-03:
You really need to step the things up. I've just finished a simple fan controller project (just starting with electronics) and I've spent half of the total time fighting Fritzing.

landracer, 2015-10-03:
The Fritzing project is not dead..... BUT it's also free... WHAT DO YOU WANT??? Please pull the source code and see what you can do to contribute to the project. NO??? WHY NOT???? See, this is more of an active approach to helping things move forward. Not this more, more, more = now, now, now BS. That's just childish. Please try again later......

duff2013, 2015-10-03:
I really wish that there was some activity here on the fritzing-app from the guys that maintain it. Maybe some guidelines or "what's needed" to facilitate advancements?

landracer, 2015-10-03:
duff2013,

Actually there has been activity by the maintainers of this project. Remember this is free software, and everyone has a life to live. Well, at least the maintainers do.... So with that said, a new forum will be implemented on the Fritzing site soon. 

Also, you can download the source code FROM GIT in the meantime and see for yourself how you can contribute(you're already here btw). There is plenty to do without instruction from the maintainers. 
Just take an active role and do something. Familiarize yourself with the code, make contributions. A move towards versioning outside of Beta. Lets not forget that Fritzing is labeled BETA for a reason. 

Hope to see some of your contributions soon!! GoodLuck!

duff2013, 2015-10-03:
Yes I know its free, I actually have contributed with hard earned cash my friend by getting pcb made from them! @landracer, are you affiliated with fritzing, I don't see any pulls from you, open or closed? Where are you getting this info from?

landracer, 2015-10-03:
I've talked with maintainers good sir. So yes, I consider myself affiliated with the project. As can anyone else. But that is aside the point. As the title of this post states, Is Fritzing actively developed? The very short answer is YES. Maybe not as some would deem fit. But it will continue to move forward. Forked or not. This project is not dead. It's just that simple. A maintainer's confirmation will make this better? Let's just count on the worst and hope for the best. and thats that! 

My information is first hand. Have faith it came from a reliable source. :)

aknoerig, 2015-12-11:
Please continue the discussion in the new forum.

closed 2015-12-11 by aknoerig