fritzing/fritzing-app #3132: Allow saving as non-zipped (fz instead of fzz) for version control?

SilverWingedSeraph, 2015-08-18:
Would it be possible to add the ability to save in a more git-friendly format? I.e., the fz XML format without the zipping.

ptaffet, 2015-08-26:
I also needed this feature, so I hacked it into the code. It's not particularly elegant, but it only took about 30 lines in FolderUtils::createZipAndSaveTo and FolderUtils::unzipTo that just mocked the zipping/unzipping and just copied the files directly without compressing.

aknoerig, 2015-08-27:
Nice, can you share the code? We could add it as a "save as...extracted fz" option.

ptaffet, 2015-08-27:
I forked it and committed these changes into my fork. My code works, but is quite ugly because I was trying to do it fast. The way the code in my branch works, there's a method at https://github.com/ptaffet/fritzing-app/blob/master/src/utils/folderutils.cpp#L345 that you can use to decide whether to create a real fzz or a fake fzz that just lists the files (then the files that would be in the fzz are saved directly in the same folder as the fake fzz). I was thinking that it could return true if the save destination is in a github repo or something like that, but to integrate this into actual production code, it should be a user decision.

zeehio, 2015-11-09:
This would be a nice to have feature!

* What should be done to convert the "ugly" code (your words, @ptaffet) :wink: into something mergeable for the next Fritzing release?
* Would @aknoerig review and accept a pull request if the patch is well written and meets quality requirements?

If @ptaffet is not available for tidying his patch I may try to to do it, although my experience with C++ and Qt code is a bit rusty/limited.

aknoerig, 2015-11-09:
@zeehio: absolutely, I'm happy to review!

ptaffet, 2015-11-09:
@zeehio My code is ugly, but worse, the way is exposed to the user is not acceptable for a production style. It should be an option in the Save As box. That requires more knowledge of Qt. I'm pretty busy this month, but feel free to use anything in my code. Two design points that need to be worked out are:
* How do you deal with the case when the zip would have multiple files? I'm not exactly sure when this happens, maybe if you include Arduino code. I solved this (IIRC) by storing in a folder.
* If the .fzz file becomes a folder, how do you open it? I solved this by keeping a "stub" fzz that caused Fritzing to look in the folder instead of unzipping the fzz to a temporary folder.

aknoerig, 2015-11-10:
Yeah, this can get tricky, so let's approach it pragmatically:

* A .fzz always only contains one .fz. Bundled with it come a couple custom part files (.svg, .fzp) and program code files (.ino, .bas,..).
It's totally fine to use the fzz saving routine and throw everything into a folder, and then just skip the zipping step. 
This is also the model that Arduino and Processing are using. We used to do that, too, but got too many complaints of people who were missing parts.

* Opening can simply done by opening the .fz file that sits in the folder -- no need to create a stub fzz. Currently, the user will get a deprecation warning in that case. I think that can be removed altogether, since we switched to .fzz a looong time ago. Alternatively, the warning could be adapted to the new scenario.

zeehio, 2015-12-13:
I just submitted a first pull request that allows to save and load uncompressed fz files. I am not fully satisfied with the proposed solution (I think it should be extended to custom parts and other bundled file types) but it seems a good starting point for getting some feedback.

As you can read on the pull request, I basically followed your suggestions, so now it should be possible to load and save uncompressed fritzing sketches.

Just a disclaimer: I have done my best, but saving and loading data is something that needs to be coded well, or data may be lost... so please be thorough when reviewing the code, as I may not be aware of all the possible code paths or I may have misunderstood something. I am doing this on my free time and I can't be hold responsible of data loss or other issues :-S

aknoerig, 2016-06-02:
No issues found with this by now, so it will be released with 0.9.3. Thanks again!

closed 2016-06-02 by aknoerig