fritzing/fritzing-app #3135: Crash on routing request

Sagittarius21, 2015-09-16:
Nom d’événement de problème:	APPCRASH
  Nom de l’application:	Fritzing.exe
  Version de l’application:	0.9.2.0
  Horodatage de l’application:	55083506
  Nom du module par défaut:	ntdll.dll
  Version du module par défaut:	6.1.7600.16385
  Horodateur du module par défaut:	4a5be02b
  Code de l’exception:	c0000005
  Décalage de l’exception:	0000000000051da0
  Version du système:	6.1.7600.2.0.0.256.48
  Identificateur de paramètres régionaux:	1036
  Information supplémentaire n° 1:	21d9
  Information supplémentaire n° 2:	21d9a2c45f8e00851059440e9291d9ac
  Information supplémentaire n° 3:	42a7
  Information supplémentaire n° 4:	42a767648eb2f641e37116fc742c443b

[image]

aknoerig, 2015-12-10:
Can you share the actual file here, too, please? Otherwise we cannot track down the cause of the crash.

KjellMorgenstern, 2019-07-12:
This issue was closed due to missing information.
If you think this issue is still important, please provide additional information.

closed 2019-07-12 by KjellMorgenstern