fritzing/fritzing-app #3138: Make a part with more than 128 pins

wan35fr, 2015-09-30:
We are trying to create a part with more than 128 pins for B Discovery Kits / D Discovery kits (PowerPC automotive)
Is it possible to extend to 256 pins ?
some microcontrollers (32 bits) begin to have more than 128 pins.

          Best regards

                                    Erwan

landracer, 2015-10-01:
Sure, anything is possible... Where is the data-sheet? Have you had a go at making it yourself??? If so, how are you trying to create such a masterpiece?

wan35fr, 2015-10-02:
Hello ,

About Datasheets , (Cf links below)
1) for D-Discovery , 128 pins are sufficient for us  ..
http://www.st.com/web/catalog/tools/FM116/CL1620/SC959/SS1675/PF259175

2) for B-Discovery,  cf link in ST Microelectronics .. we need 256 pins max
http://www.st.com/web/en/catalog/tools/PF259176

3) for P-Discovery , (With Arduino shield) .. we need 256 pins max
http://www.st.com/web/catalog/tools/FM116/SC959/SS1675/PF259567

i do not receive a "go" to update Frizing-App into 256 pins.

Anyway, for D-Discovery , we have succeeded to create a Fritzing component by following some useful tutorials on Internet.
we will commit our work on Fritzing-Parts
Fritzing is clearly a convenient tool to create some small projects on our "small" community.

                   Best Regards

                                     Erwan

landracer, 2015-10-04:
Thank you Erwan! Anyone making parts helps move the project forward :+1:

aknoerig, 2015-12-10:
Hi Erwan, where did you find the limit to 128 pins?

KjellMorgenstern, 2019-12-04:
@microwan It seems this was just a question, not an actual problem?

Closing because no further information was given.

closed 2019-12-04 by KjellMorgenstern