fritzing/fritzing-app #3139: UI too small on Windows 4K monitor; does not respect Windows Display Size (scale) parameter.

softegg, 2015-10-07:
The Icons and other UI elements are just too small to use on a 15" 4K monitor. Normally, windows can tell programs that it needs a higher scale, but Fritzing doesn't seem to support that.
A work around is to set up a manifest file and set windows to always use the size parameter from there.

rennex, 2015-10-25:
Same issue here on a much larger monitor which is still about 150 DPI. I'm just barely able to use the program when I lean in really close. It even has that "supports HiDPI" flag set in the .exe, so Windows doesn't let me scale it bigger even if I wanted to.

The window menu and Preferences dialog are large enough, but almost everything else is tiny. They need to be able to scale to 1.5x or 2x the current size in order to be usable on new high-DPI monitors.

aknoerig, 2015-12-19:
Seems like an upgrade to Qt 5.6 could help: http://blog.qt.io/blog/2015/12/18/qt-5-6-beta-released/

It's only in beta though, so we still need to wait.

aknoerig, 2016-03-20:
Qt 5.6 is finally out, so this is fixed with enabling scaling in commit 1dfb8cf5b1654590d6f063b3e25a3670c8b441af

closed 2016-03-20 by aknoerig