fritzing/fritzing-app #3140: impossible to enter decimal figure when locale uses comma instead of decimal point

tmlmt, 2015-10-09:
My config: Xubuntu 14.04.3 LTS 32 bits, Fritzing 0.9.2b

When, in my linux system, Settings > Language Support > Regional format is set to some place (e.g. France) where a comma is used instead of the decimal point to delimit decimal numbers, it is impossible to set a part's size with a decimal number. It seems that the field's character filter reject attempts to add commas. 

=> as done in calculator apps for instance, Fritzing should detect when the "," (main part of the keyboard) or "." (numpad) is pushed and interpret it automatically as a decimal point. 

I changed the regional format to "English (US)", logged out and in again and it worked. 

Tested for instance with the generic PCB hole part, which has two settings for hole size.

alexfomenko, 2015-12-12:
I had the same problem on versions greater than 0.9.0b. I'm on a MacBook 7.1 OS 10.9.5, and I'm from Russia. Year ago I simply rolled back to Fritzing v0.8.7b. But thank you, @tmlmt , now I can use newest versions with no problems, and without changing my region and even relogging! 
So on a Mac you need to go to System Preferences > Language&Region > Advanced > then select "." in number separators and currency for any case. Restart your Fritzing and done!
[image]

4C3T, 2018-10-06:
Hi. I'm using Windows 10, and it'd seem like regional settings is causing this for me as well. I can't input correct numerical values. I can't either (copy/)paste nor write dot in a numerical field, making routing impossible for through-holes, so it's quite useless as is.

Do I really need to change my regional setting in order to bypass this???
Earliest post above is since 2015, yet no fix for this.. Come on..!

failiz, 2021-03-12:
actually, I think that it is a duplicate of #3080. Lets close it and keep the other one.

KjellMorgenstern, 2021-03-12:
Closing in favor of #3080

closed 2021-03-12 by KjellMorgenstern