fritzing/fritzing-app #3141: Make note font size changeable; note fonts are HUGE compared to parts labels

ElectricRCAircraftGuy, 2015-10-10:
Please look at the part label font sizes (ex: D13/SCK, D12/MISO, etc) in the below image, as compared to the note font sizes. It's pretty absurd. In order to blow up my schematic enough to actually make it readable when I want to print it, I have to first delete all my notes. This could be easily fixed if I could simply make the note fonts as small as the parts labels, as that would *dramatically* reduce the amount of space each note takes up.

Thanks!

[image]

ElectricRCAircraftGuy, 2015-10-10:
One more thing: please allow a PDF and Image export option to make all parts fonts black instead of grey. This would help greatly when printing the schematics too, so that the printer doesn't try to print with black ink sparsely to make gray. Printing the dots that make up the words sparsely simply makes them unreadable. If they were black, the printer would print them higher resolution and they would be readable.

aknoerig, 2015-12-10:
I can't replicate this behaviour on OSX here -- did you maybe paste the note text from an external application? If not, can you give precise steps how to replicate this issue?

As a temporary workaround for comment 1, you could export to SVG, then use Inkscape or Illustrator to set everything to black.

ElectricRCAircraftGuy, 2016-01-13:
Hi, to replicate the large note fonts, I literally just open Fritzing, open any Fritzing file or create a new one, and insert a note (by clicking "Add a note") while *in schematic view*. In Schematic view, all notes I insert have huge font. In Breadboard view, the note font is smaller.

ElectricRCAircraftGuy, 2016-01-13:
Here is breadboard view, just after creating a new file, dragging out a diode, then clicking "Add a note":

Notice the note font is reasonably sized compared to the diode and the word "fritzing" in the bottom left.

[image]

Here is Schematic view, after clicking "Add a note." Notice how *tiny* the diode part font is compared to the comparatively *huge* note font. We need the ability to change the note font size. I like to write entire paragraph descriptions sometimes, and the notes take up waaaaaay too much room.

[image]

ElectricRCAircraftGuy, 2016-01-13:
Here's a practical, real-life example. I just finished this circuit. I added some important notes at the bottom, and the notes are so HUGE I can't even print this circuit for reference. I have to zoom out so far to allow the notes to fit that now I can't read anything. :(

[image]

ElectricRCAircraftGuy, 2018-09-23:
Kev, I found my preferred solution: I switched to KiCad.

KjellMorgenstern, 2020-12-06:
@val43 Github has a feature to "+1" issues, please make use of that. The +1 comments are generally not helpful for statistics.

KjellMorgenstern, 2024-03-10:
There are still problems with font in Notes not properly handled. Basically, Fritzing is not aware of richtext content (from copy and pasting) and does not handle it correct in all cases.