fritzing/fritzing-app #3145: cannot get tactile button to press on

cardac, 2015-10-14:
hello - I don't know who is messing around.. but now I cannot get my tactile switch to PRESS ON..!!!

has something changed.... before I could do that....

how do we do it now... (I built a simple circuit on breadboard to show someone... and now I cannot get the tactile switch to "click on"....!!! so that the LED will light up...

thanks....

landracer, 2015-10-14:
Fritzing does NOT do circuit simulation.

So in short, no one is messing around. Not sure what you're doing...... If you want a led to lightup built it on a bench breadboard. The breadboard feature in fritzing is for helping you build a circuit, by allowing you to replicate what you've already done on a bench.

Sorry if that's not how you remember things. But that's how it works. I could be completely wrong. But I don't use much of this side of the program. I just know there is no, 'on switch'

Good luck!

cardac, 2015-10-15:
hi there... buddy... thank you very much for answering my mail..!!!

I like Fritzing very much..!! and use it to show disadvantaged kids .. lovely things...!! they love it...! !!

I would like to donate... but from South Africa.. I can only use a credit card.. NOT PAYPAL.!!

Let me know..!!

Regards
Spencer..!!

closed 2015-12-10 by aknoerig