fritzing/fritzing-app #3146: Unable to find 0.1" break-away headers

chrisspen, 2015-10-14:
Searching for 0.1 break way headers finds several results, many that specifically describe themselves as such. However, when I try to place the part (e.g. "Header 3") it only renders as a "screw lock" variant with a "screwterminal-3.5mm-3lock" package. The variant and package drop-downs are broken on Linux and hide half the items and show the selected item as white text on a light gray background, but I can't find any other combination of values that makes this part render as a regular 0.1" break-away header.

Is this a bug or am I missing something?

I'm using version 0.9.2 on Ubuntu 14.

landracer, 2015-10-14:
In Fritzing version 0.9.2. I search 'header', then scroll to the bottom of parts list. there is pin header. Icon is like a single black hole. Select this, then in the parts dialog box change your parameters as necessary. Everything is fine here on my end. I use headers all the time with varying amounts of pins, never had an issue.

I'm running fine on kernel 3.13.0-38-generic. I'm am not using regular Ubuntu though, but a variant. 
Are you compiled from source?  

Install fritzing from the ubuntu repo. This way we can see if you still have the same issues. Issue these quick commands in a terminal.
sudo apt-get update
sudo apt-get build-dep fritzing
sudo apt-get install fritzing

chrisspen, 2015-10-14:
Ah, perfect. That's what I was looking for, thanks.

I'm using the pre-compiled version. I was using the one from the Ubuntu repo, but that was only 0.8.5, which I thought was a little old.

closed 2015-10-14 by chrisspen