fritzing/fritzing-app #3148: Unable to set PCB to single-sided copper

chrisspen, 2015-10-15:
My circuit is able to fit on a single side, but when viewing it in the PCB view, the auto-router seems to split it between sides, and there doesn't seem to be any option set it to only use a single side.

Are single-sided PCBs supported?

landracer, 2015-10-16:
You might be letting Fritzing work with both layers. In PCB view at the bottom of the screen click on both layers till it says TOP LAYER, then try again. This should keep the auto router working on the top side surface.

Good luck

chrisspen, 2015-10-17:
That only makes the top layer clickable, but it still keeps both layers. Even after it says TOP LAYER, running auto route still routes on two sides.

landracer, 2015-10-17:
Turn off the bottom in view mode(forgot that part last time). I know this works. I use it for fill purposes all the time. As time goes I'm sure usability of this feature and more layers will progress. The software is still currently in BETA. developments have slowed. So... Have a play around. I know it works with copper fill. But most of the time my circuits are to complex for the autoroute feature. Which is not the best working part of this program anyhow. Don't be afraid to drag a few wires yourself and make it look as you
please.

chrisspen, 2015-10-17:
I figured it out. If I select the PCB then I can set the layers property to "one layer (single sided)". Then Autoroute works as expected.

Thanks for your help.

closed 2015-10-17 by chrisspen