fritzing/fritzing-app #3151: License question

improti, 2015-10-17:
I just stumbled upon a paragraph in the main readme.md
> You may publish circuits and diagrams that you create with Fritzing and that use our graphics, again as long as you credit us, and publish your works under the same license.

For as I understood the GPL the output of the program is not covered by the license, i.e. you do not need to make generated output available. I read about that in a discussion of gcc and it's output (i.e. the compiled programs). Is the difference here because of the graphics used and the license associated with those? Or is this maybe a difference between v2 and v3 of the GPL? Might be my info is about the v2 and fritzing-app is using v3.

Just to clarify: I ask this out of curiosity and because I try to understand the licensing schemes. This wouldn't effect me as I intend to publish my results anyway :)

landracer, 2015-10-20:
I'm not the one to really speak on this one. So lets for instance

Using pictures of the program with your said 'project' and cropping out Fritzing aka no credit. Just throw some credit where credit is due, when that said time comes. Even if its a simple "I used Fritzing". :)

If you're on the level of making proprietary circuitry, no you don't have to publish your said "files" 

When do you have to publish your 'files'? When claiming OSHW, then you better be doing what you need to do... Post your files. Preferably all of them, not just some schematic and gerber files.

You can also use a gerber viewer and post as you please giving no credit to anyone... Happens all the time, aside who is going to come slap anyones hands if you go this route??? NO ONE.

It's all about giving back to the community in some way or another. So moral of the story, give credit where credit is due. GO Fritzing VER 1.0 OMG! <<<< That would be, um not Beta anymore???? Well its down the pipe somewhere I'm sure...

improti, 2015-10-20:
As said, no problem with me. But it sounded as this would be a requirement license-wise, which I just assume it is not. You do not have to write "compiled with gcc" into your program, for example. A better analogy might be that you do not have to write "written with LibreOffice" under your letters. Oh, LibreOffice is not GPL, but ...

niclashoyer, 2015-12-05:
The GPL is just used for the code, so it regulates modifications of the code and distribution of compiled binaries. It does not impose anything on the output produced by Fritzing.

The parts you use in Fritzing are licensed as [CC-BY-SA 3.0](https://creativecommons.org/licenses/by-sa/3.0/), so if you are using these parts in derivative works (such as schematics or pcb layouts) you'll have to comply with the license. If you share any derivative work you must credit the author and also use the same or a [compatible license](https://creativecommons.org/compatiblelicenses).

improti, 2015-12-22:
Sounds right. But composing a list of all authors sounds tedious. Is there any functionality to provide a list of the authors of all parts used? If not, make this a feature suggestion. The needed information should be available in the libraries, right?

I'd like to publish some things on a web-site and wonder how to do it right. Any help appreciated.

improti, 2016-01-19:
What's needed is machine-readable information on contributors and licenses of:
* Fritzing itself
* All components used in the current project

The function should provide a text-only (or maybe HTML in addition?) output that can easily be included in copyright notices. It should contain:
* list of "projects", i.e. at least Fritzing + any used component-libraries
* list of licenses used by each involved project
* list of contributors that need to be mentioned when publishing the work

I haven't done any work on the code yet, but if someone could point out if/where the information is stored, I might actually be able to compose a report (by modifying a copy of let's say the component-list-report) doing this. Any thoughts/hints?

RobertGallen, 2022-09-05:
Just been debating this question myself with colleagues, so was pleased to find this issue. The FAQ is still inconsistent with the conclusion of this thread and also the licensing section of the Readme. For what it's worth, I think the FAQ is the one which is incorrect and needs updating/caveating. 

Note that although I'm posting from a commercial account, this is for an academic project where CC-BY-SA would not necessarily be an issue.

KjellMorgenstern, 2022-09-05:
Disclaimer: I am the maintainer of this project. I am not a lawyer. Please be aware, that this GitHub issue is an open discussion about improvement and communication of licensing for Fritzing. It is not a change in our licensing or anything effective. 

So.. well, this is quite a complex topic, with a lot of pitfalls and misunderstandings. With the potential to eat up months of work. In general, I agree with the short summaries from @landracer and @niclashoyer 

To address the original topic of the issue from @improti 
I think managing such a list is not in the scope of Fritzing. Managing such a list applies to all kind of situations in a project, by far not limited to Fritzing parts. If you publish something, it is your (often academic) duty to maintain references. Similar applies for commercial projects. Everyone should have their strategy to manage references, and a partially automated system will be of very limited use. Simple modifications are often too low in complexity to justify a copyright (and I think here already could be a point where EU and US copyright differ). On the other hand, there are companies who patent a certain curvature of a rounded box, and send a legion of lawyers after this. While we try to do our best, in the past some "creative" people even managed to upload blockbuster movies to the Fritzing project gallery, although that was a completely different (unmaintained) time for Fritzing. Don't ask me what license they had chosen for this and who was the 'author'. My point is, we can not tackle this problem with sophisticated, detailed solutions. 

So what we can do, in the sense of Open Source Hardware, the best effort, no guarantees?
We keep components included in Fritzing free from surprises. That means: CC-BY-SA for the breadboard, to Fritzing. And do whatever you want for schematic and PCB illustrations. "Do whatever you want" was probably the year 2009 term for "CC-0", but I let me know if that is wrong, maybe that is a point where the FAQ can be improved?

Do the sane thing. If someone on their website provides a Fritzing part, without explict license, contact them before using it in a mission critical project.
Don't snip the Intel logo from the Eddison board and start mass-producing "Intel" boards, just because it is possible to grab it from Fritzing, and we said do whatever you want. Our CC-0 would not protect you, the brand registration overrules that, you would be in trouble. Btw, Arduino has a community logo for such use cases (and a googleable website with explanations)

@RobertGallen Can you point out where you think the README conflicts with the FAQ?

RobertGallen, 2022-09-06:
To be completely clear, I am not pushing for a change in philosophy or licensing, I am just trying to understand the _status quo_. This is partly because we have to get this software through corporate lawyers, and while "do the sane thing" is an admirable and rational philosophy, it probably won't satisfy them. I am 100% aware that this is my problem, not yours. 

With that said, the FAQ says:

> First of all, you can do whatever you want with them. Only when you decide to publish images of the breadboard view on a website or in a book, we ask you to attribute us. Attribution can be as simple as "this image was created with Fritzing." And if you think it turned out particularly nice, drop us a link/copy!

This sounds like CC0 (do whatever you want) or CC-BY (do what you want but attribute). But the README says:

> You may publish circuits and diagrams that you create with Fritzing and that use our graphics, again as long as you credit us, and publish your works under the same license.

Which sounds like CC-BY-SA as described well by @landracer and @niclashoyer. The licence files suggest CC-BY-SA, and the readme description is consistent with that (because any diagrams created using the graphics are adapted works and so will need to be CC-BY-SA or compatible licensed. So I think the FAQ should be updated with wording similar to that currently in the README.

KjellMorgenstern, 2022-09-06:
You are right the README and the FAQ contradict, although the README itself links to the FAQ for details. I am convinced the intent was to protect the parts themselves, but still allow users to use their circuits without limitations. The corrections can be seen in the history of the FAQ, very likely going back to users who point out uses cases and problems with it.

So the wording which is on the FAQ since about 12 years, must be formulated more readable. How about the below? To some, this text might look familiar, see the KiCad licenses. I added an attribution for publishing, trying to match what we have in our FAQ. 

>__Warning__ This is just a first draft. It is unclear if we will apply it.
> 
> **Part License**
> The Fritzing parts are licensed under the [Creative Commons CC-BY-SA 4.0 License](https://creativecommons.org/licenses/by-sa/4.0/legalcode), with the following exception:
> 
> To the extent that the creation of electronic designs that use 'Licensed Material' can be considered to be 'Adapted Material', then the copyright holder waives article 3 of the license with respect to these designs and any generated files which use data provided as part of the 'Licensed Material'.
> 
> **What does this mean?**
> 
> Fritzing parts are licensed in such a way to ensure free use of Fritzing parts for commercial, closed, and non-commercial projects without restriction. Fritzing does not wish to exert any control over designs produced using Fritzing parts, or force users to reveal proprietary information contained in their designs. Neither do we wish to force users to attribute the Fritzing parts within their design. If, and only if, you publish an breadboard illustration derived from Fritzing parts, we kindly ask you to attribute it according to CC-BY 4.0.
> 
> Use of the Fritzing part data in a project does not (by itself) require that the design or any files generated from the design are licensed under the CC-BY-SA 4.0 License. You are free to use the Fritzing part data in your own projects without the obligation to share your project files under this or any other license agreement.
> 
> However, if you wish to redistribute the Fritzing parts, or parts thereof (including in modified form) as a collection then the exception above does not apply. Redistributed Fritzing part collections must be shared under the same license agreement. Under these circumstances, the Fritzing parts must also retain attribution information, including the license documents which are distributed with the Fritzing part files.
> 
> **Warranty**
> 
> Fritzing parts are provided in the hope that they will be useful, but are provided without warranty of any kind, express or implied.
> 
> **Community**
> 
> The Fritzing parts are compiled by Fritzing, with huge support from the community - if you find an error in the library data, please help the community and report it or contribute a fix! 
> 
> 
>

KjellMorgenstern, 2022-09-06:
Addition
>**Brands**
>Fritzing parts contain protected symbols, like logos and trademarks. Any usage of these protected symbols must be cleared with their respective owners. Fritzing does not, and can not, grand any permission about those protected symbols. 

Explanation
Fritzing has agreements with these companies. From some companies, community logos are available, which are possible to integrate. Most companies will allow the usage for documentation purposes, as long as the symbols are kept in context, but a permission is required nonetheless. Schematic view and PCB view should not contain any such off-hands symbols. Please let us know if otherwise, also for community parts which are not compiled by Fritzing.

For example, the Arduino logo is one of those symbols. See their guidelines here
https://support.arduino.cc/hc/en-us/articles/4679102084892-Arduino-and-Community-logos

RobertGallen, 2022-09-09:
Thanks @KjellMorgenstern, that's some useful clarification, and the project's intent is now clear. I am not a lawyer either, so I cannot say whether or not the existing licensing set up is compatible with that intent, but updating the FAQ in that manner would be helpful.