fritzing/fritzing-app #3152: Building Fritzing on FreeBSD 10.2-RELEASE-p6 fails when using system default compiler (clang++)

en90, 2015-10-28:
Hello,

I attempted to build fritzing-app on FreeBSD 10.2-RELEASE-p6.

I installed qt5 and the boost libraries.

```bash
sudo pkg install qt5 boost-all
```

Next I try to build it based on [the Linux notes](https://github.com/fritzing/fritzing-app/wiki/1.3-Linux-notes):

```bash
git clone --recursive https://github.com/fritzing/fritzing-app
cd fritzing-app/parts
git submodule update --init --recursive
git checkout master
cd ..

qmake PREFIX=/opt
make
```

The build starts but fails soon after it started.

Last few lines of output:

```
clang++ -c -pipe -O2 -Wall -W -pthread -D_THREAD_SAFE -fPIE -DLINUX_32 -DDATADIR=\"/opt/share\" -DPKGDATADIR=\"/opt/share/fritzing\" -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_XML_LIB -DQT_NETWORK_LIB -DQT_SQL_LIB -DQT_CONCURRENT_LIB -DQT_SERIALPORT_LIB -DQT_CORE_LIB -I. -isystem /usr/local/include/qt5 -isystem /usr/local/include/qt5/QtSvg -isystem /usr/local/include/qt5/QtPrintSupport -isystem /usr/local/include/qt5/QtWidgets -isystem /usr/local/include/qt5/QtGui -isystem /usr/local/include/qt5/QtXml -isystem /usr/local/include/qt5/QtNetwork -isystem /usr/local/include/qt5/QtSql -isystem /usr/local/include/qt5/QtConcurrent -isystem /usr/local/include/qt5/QtSerialPort -isystem /usr/local/include/qt5/QtCore -Irelease -I. -I/usr/local/lib/qt5/mkspecs/freebsd-clang -o release/graphutils.o src/utils/graphutils.cpp
src/utils/graphutils.cpp:35:10: fatal error: 'boost/config.hpp' file not found
#include <boost/config.hpp>
         ^
1 error generated.
*** Error code 1

Stop.
make[1]: stopped in /usr/home/erikn/src/github.com/fritzing/fritzing-app
*** Error code 1

Stop.
make: stopped in /usr/home/erikn/src/github.com/fritzing/fritzing-app
```

Next, I tried using GNU tools instead

```bash
# Since there are neither "clean" nor "distclean" targets in the generated Makefiles,
# remove everything and clone again. Could probably have used "git clean" or something.
cd ..
rm -rf fritzing-app

git clone --recursive https://github.com/fritzing/fritzing-app
cd fritzing-app/parts
git submodule update --init --recursive
git checkout master
cd ..

QMAKESPEC=freebsd-g++ qmake PREFIX=/opt
gmake CXX=g++48 CC=gcc48 LINK=g++48
```

That works and I'm then able to

```bash
sudo gmake install
```

Well, almost -- I had to remove a GNU-specific flag (a flag which turned out to not be necessary either way) to the icon `install` command in `phoenix.pro`.

```patch
diff --git a/phoenix.pro b/phoenix.pro
index 0672692..4fe3128 100644
--- a/phoenix.pro
+++ b/phoenix.pro
@@ -118,7 +118,7 @@ unix {
         manpage.files += Fritzing.1

         icon.path = $$DATADIR/icons
-        icon.extra = install -D -m 0644 $$PWD/resources/images/fritzing_icon.png $(INSTALL_ROOT)$$DATADIR/icons/fritzing.png
+        icon.extra = install -m 0644 $$PWD/resources/images/fritzing_icon.png $(INSTALL_ROOT)$$DATADIR/icons/fritzing.png

         parts.path = $$PKGDATADIR
         parts.files += parts
```

```bash
QMAKESPEC=freebsd-g++ qmake PREFIX=/opt
sudo gmake install
```

Once built and installed, `Fritzing` segfaults when I try to run it but that's a separate issue and might be just because I didn't have an X11 session though then again on Linux it actually says "cannot connect to X server" when such is the case so who knows.

aknoerig, 2015-12-10:
Sorry, I don't have a machine to test this on. However, Fritzing definitely requires a GUI to run, it cannot run headless. It inherits from QApplication (http://doc.qt.io/qt-5/qapplication.html#details).

nopdotcom, 2017-12-23:
Fritzing's build process can run headless on the Mac with one change, #3339.

KjellMorgenstern, 2019-04-30:
This issue was closed because it is outdated or solved.
If you think this issue is still important for a reason, please open.

closed 2019-04-30 by KjellMorgenstern