fritzing/fritzing-app #3157: Export pcb in 3D

b4zz4, 2015-11-13:
I think that this serious very useful to create boxes and supports.  I think that very well use openscad.

htho, 2015-11-16:
Hi,

you may take look at my projects https://github.com/htho/fzz2scad and https://github.com/htho/scadlib

Best Regards
htho

b4zz4, 2015-11-16:
Thank you very much :D

BrainsoftLtd, 2016-06-03:
I was unaware of this existing issue, and have created #3219 to present an alternative approach to creating a 3D model using the Eagle3D part library and POV-Ray.

failiz, 2024-01-09:
Duplicate of https://github.com/fritzing/fritzing-app/issues/4059
Current workaround from #4059: "Please try Fritzings Gerber and IPC export together with ZofzPCB. This will produce a 3d model of a Fritzing PCB, and it provides a basic editor to add parts that don't have a rendering. To get good results, you will need to buy ZofzPCB, but there is a free evaluation period."