fritzing/fritzing-app #3159: convert wire to jumper

b4zz4, 2015-11-18:
[image]
[image]

htho, 2015-11-18:
Is this a feature request? I support it!
A first easement would be a jumper that can be changed in length similar to resistors. Diodes should allow different lengths as well.

My current work-around (when designing one-sided perfboards in PCB-View) is to "define" the other side of the PCB as jumper. But this is a very limited approach.

b4zz4, 2015-11-18:
I was wishing what might be replaced the cable for the bridge. With a simple option of the menu

aknoerig, 2015-11-19:
Thanks, good idea.
Note that it's already possbile to change the length of the jumper wire. It's not very obvious, but you can actually simply grab one end of it and drag it–you just need to click outside of the copper ring.

htho, 2015-11-19:
I did not know that. The amount of time this knowledge could have saved me...

Anyway, the "replace trace with jumper" function would be great.

b4zz4, 2015-11-19:
André Knörig:
> Thanks, good idea.
> Note that it's already possbile to change the length of the jumper wire. It's not very obvious, but you can actually simply grab one end of it and drag it–you just need to click outside of the copper ring.

Yes, i use this

h2CSO, 2017-06-08:
I support this feature too!

nando2301, 2020-04-14:
I support this feature too!