fritzing/fritzing-app #3163: Ruler: 1/10 millimeter scale

duff2013, 2015-12-07:
it would be nice to have a finer scale for the ruler. Is this something that would be considered for a pull?
[image]

niclashoyer, 2015-12-07:
yes, please, pull requests are always welcome!

duff2013, 2015-12-07:
Ok, I was also thinking of splitting the ruler into just metric or english depending on what width type property was selected. The ruler in my opinion is unruly in such that its too wide while displaying Inches and Centimeter together and if I choose cm from the properties drop down I would think users just want to use cm for the ruler not both, same goes for if I choose inches. 

In the long run I want to look at a proper measuring system maybe like a simplified SolidWorks measuring tool which can select edges and measure the distance between them. So for example I could click the edge of the pcb and the edge of part and get that distance. I know it won't be easy or even possible but would make placing components and such much more precise.

duff2013, 2015-12-09:
here is what I have come up with for the Ruler Class:
[image]

As you can see it splits the ruler into its own unit type i.e. English or Metric. Also made switching these types dynamic so when you switch units the Ruler changes to reflect that change. I am looking into fixing that stupid ComboBox that Qt has becuase it is a real pain to select the option you want. I was thinking of changing it to simple Check Box type?

aknoerig, 2015-12-10:
Nice work! Not sure about Check Box, but a radio button would also be appropriate here.

duff2013, 2015-12-10:
Thanks! I'll have a pull I hope today, I'll do separate pull for the radio button, still learning the fritzing code flow so I want be sure I don't break anything. My changes to the Ruler Class should be localized to that class so hopefully its incapsulated correctly:)

aknoerig, 2015-12-14:
Committed in PR #3166

closed 2015-12-14 by aknoerig