fritzing/fritzing-app #3165: export / save my_parts bin

FritzWare, 2015-12-09:
I'm using fritzing.0.9.2b.64.pc in windows 7

I exported and/or saved (I'm not sure) the My_Parts bin under the same name in several places. After starting Fritzing again I see copies of the My_Parts bin.
Deleting the exported bins have no effect.
Deleting the Roaming bin and Cleaning the PC-Registers have no effect.
Restart the computer and using a complete new version of Fritzing have no effect.
So I think there there must be some register that my clean up program does not clean.
At this moment I'm working with 6 copies of the My_Parts bin.

How to delete these copies?

Fritz

aknoerig, 2015-12-10:
Hmm, all user-specific Fritzing stuff should actually be in the Roaming folder.  See the table at https://github.com/fritzing/fritzing-app/wiki/2.1-Part-file-format#folder-structure -- can you confirm you checked this specific one?

FritzWare, 2015-12-11:
André,

With help of a PC expert it was found that my PC had multiple copies of the my_parts bin in the registers.
After deleting them all the problem was cleared.

Then I tried to reproduce the problem with reproduceable success.

"Saving as" the my_parts bin repeatedly one extra bin in fritzing.
The used name and place makes no difference.
This extra bin can be deleted in fritzing.
Deleting the file will also delete the bin in fritzing.

More problematic is the use of the export function.
Every "export bin" of (only) the Original my_parts bin leads to an extra my_parts bin.
The used name and place makes no difference.
The bin has no delete function in fritzing and deleting the fzbz file has no effect.
What can be done: use the "save as" function for all copies.
Then the delete function comes available for all copies except the original my_parts bin.
Another way is to find and delete the corresponding pc registers (regedit + find fritzing).

So with this work around my problem is manageable.
When reproducable on other computers it might need a fix.

Frits

aknoerig, 2015-12-11:
Thanks for digging deeper into this. I'm surprised there's anything showing up in the registry though, since we're not actively registering anything there.

closed 2015-12-11 by aknoerig

FritzWare, 2015-12-11:
André,

I forgot to mention that saving and exporting the Core bin show the same problems as with the My_parts bin.

Frits Damen

Closed #3165.—