fritzing/fritzing-app #3172: Bending lead on breadboard breaks schematic connection

mMerlin, 2015-12-21:
It appears that bending a lead on the breadboard view *always* ? disconnects that connection point from anything it is attached to in the schematic view.

Environment
x86_64 Windows 7 Professional
Fritzing 0.9.2 (b11.19) 64 (b8d2d5970658f…)

issue #2950 says that straightening a bend can sometimes cause a disconnect.  I found that bending the leg of an LED on the breadboard view (to **make** a connection) breaks the connection on the schematic view.

Very simple test circuit.

NoBend.fzz
On schematic view, connect 5V power label, THT resistor, THT LED, and ground symbol in series with 3 wires.  On the breadboard view
* place the resistor from the plus (red) line of the breadboard to column "I"
* place the LED so that the anode is in the row adjacent to the resistor

At this point, the schematic view shows 'routing complete', and the breadboard view shows 1 connection to be routed.

BendIt.fzz

* on the breadboard view, bend the anode lead of the LED to bring it to the row with the resistor lead.

Breadboard view shows Routing complete, but now schematic view shows 1 connection to be routed, and the supposedly joined ends of the resistor and LED are red, showing no connection.

MoveBroken.fzz
* on the schematic view, move the resistor and LED.  The end of the connection wire moves with the resistor, but not with the LED, and a rats nest wire line is displayed to connect the resistor and LED legs.

FixBend.fzz
* move the end of the wire in the schematic to reconnect it to the LED

Both schematic and breadboard view now show routing complete.

UnBend.fzz
* on the breadboard view, bend the anode lead of the LED to bring it back to the original row.

Breadboard view says there are no connection to route, no rats nest line exists.
Schematic view says routing is complete, but the resistor and LED are again not connected.

MoveAgain.fzz
* on the schematic view, move the resistor and LED again.  As before, the end of the connection wire moves with the resistor, but not with the LED.  This time though, no rats nest wire line connects them.

.fzz files renamed to .zip, so github would let me attach them here.

[BendIt.zip](https://github.com/fritzing/fritzing-app/files/69093/BendIt.zip)
[FixBend.zip](https://github.com/fritzing/fritzing-app/files/69095/FixBend.zip)
[MoveAgain.zip](https://github.com/fritzing/fritzing-app/files/69097/MoveAgain.zip)
[MoveBroken.zip](https://github.com/fritzing/fritzing-app/files/69094/MoveBroken.zip)
[NoBend.zip](https://github.com/fritzing/fritzing-app/files/69096/NoBend.zip)
[UnBend.zip](https://github.com/fritzing/fritzing-app/files/69098/UnBend.zip)

failiz, 2021-10-07:
I could not replicate this in 0.9.9. @mMerlin , can you still replicate this?

KjellMorgenstern, 2021-12-09:
The leg disconnecting from the part (#2950) is not a duplicate.

KjellMorgenstern, 2021-12-29:
We can't reproduce this in 0.9.9, closing for now

closed 2021-12-29 by KjellMorgenstern